MarlinFirmware/Marlin #10474: [bugfix-2.0.x] Fade height problem

letgo04, 2018-04-21:
I tried most recent verions around these days. now I tested with 17.04
platform RADDS with tmc2100.

1. Setting MBL then z-offset around -0.1 in LCD controller and saved 
2. Without setting fade height, MBL works well. no problem.
3. Set fade height to 2

**Expected behavior:**
 till 2mm z axis should move for compensation. 

**Actual behavior:**
yes. from 2mm, it doesn't move for compensation, but from first layer to 2mm layer whenever moving next layer, it looks like z jumping around 1mm?(I couldn't measuer exact height) By the way in LCD controller, z height looks normal.
And during printing, when adjusting fade height from 2 to 0, in contrary z moved to nozzle as height as it jumped before(it is just my guess, when I see it as my observation)

there is no problem without fade height till now.

#### Additional Information

I have tried to debug this for around week. and what I found out is that

planner.h

      inline static float fade_scaling_factor_for_z(const float &rz) {
        static float z_fade_factor = 1.0;
        if (z_fade_height) {
          if (rz >= z_fade_height) return 0.0;
          if (last_fade_z != rz) {
            last_fade_z = rz;
            z_fade_factor = 1.0 - rz * inverse_z_fade_height;
          }
          return z_fade_factor;
        }
        return 1.0;
      }

when I changed
 z_fade_factor = 1.0 - rz * inverse_z_fade_height;

to

 z_fade_factor = some const number like 0.5 

then there is no problem. it works as expected as I am.

thinkyhead, 2018-04-22:
> - "from first layer to 2mm layer … z jumping around"
> - "without fade height or with above 1, it looks there is no z-offset problem."

So… I'm confused. Are you setting it to 2 and seeing problems at that value, or are you setting it to 0.2 and that's when it exhibits problems?

letgo04, 2018-04-22:
problem exist whenever I set fade height.
I tested simply by using lcd controller..
tested fade height 1 and 10. 
1. reboot print
2. auto home
3. after finished auto home, turn on bed leveling in lcd menu
4.  set fade height
5.  when I use move_z(in lcd controller) till fade height,  I saw it moves actually 5mm(when fade height set 1 and I moved to 1mm) and 14mm(when fade height set 10 and I moved to 10mm)
6. over the fade height, it moves well.

letgo04, 2018-04-22:
I test it as I wrote here. by using gcode G29. modifyed MBL data and also z offset.
it looks there is no jumping z.

thinkyhead, 2018-04-24:
```cpp
get_z = 8 * 0.8 = 6.4
```
What makes you think that `z0` would ever be 8.0? The mesh values are supposed to center around 0 and typical values should all be from -0.5 to 0.5, at the extreme.

letgo04, 2018-04-24:
SENT: M420 V
READ: Mesh Bed Level data:
READ: Num X,Y: 3,3
READ: Z offset: 0.00000
READ: Measured points:
READ:         0        1        2
READ:  0 -9.58750 -9.40000 -9.60000
READ:  1 -8.92499 -9.04999 -9.12499
READ:  2 -8.74999 -8.89999 -8.72499
READ:

thinkyhead, 2018-04-24:
Hmm. You're doing something wrong there. The nozzle should be right at the bed when doing mesh measurements, and the mean value should be 0. Let's do some logging!

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed and enable leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

letgo04, 2018-04-24:
The nozzle should be right at the bed when doing mesh measurements?
even if physical distance is aroun 9mm?

thinkyhead, 2018-04-24:
Yes! When doing MBL measurement, the nozzle moves to each point in turn and then goes down to the bed. You put a piece of paper under the nozzle and adjust the nozzle up and down to the point where it just grabs the paper but you can still move it pretty easily. Then you click (or without an LCD, do `G29 S2`) to save the height value and move to the next point. Repeat until all points are saved.

letgo04, 2018-04-24:
Yes. I understand how to use MBL. by the way, with this physical distance 9mm, how nozzle and bed can meet nearly 0 distance when beginning MBL??
this 9 mm is decided by mechanic endstop of z. maybe if I adjust height of z endstop upper 9mm, maybe they could meet. but in my structure it is impossible to install endtop upper.

thinkyhead, 2018-04-30:
> this 9 mm is decided by mechanic endstop of z

Are you saying that when you home Z to the minimum endstop, the nozzle stops at 9mm above the bed? Is this a Hall effect endstop that allows the nozzle to move farther down after it triggers?

letgo04, 2018-04-30:
I modified source a bit.
after finishing MBL, I made average value of  MBL, then subtract it from each data of MBL.
finally set average value to z offset.

No problem now. Thanks.

thinkyhead, 2018-05-01:
Ok, but it sounds like your configuration is quite messed up.

boelle, 2019-02-20:
@letgo04 still having issue with latest bugfix 2.0?

if you dont have problems then maybe consider clicking the green close button below?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-17 by p3p