MarlinFirmware/Marlin #10477: DUAL Z Mode: E1(Z2) wrong direction

noxx2015, 2018-04-21:
Hello

i use Marlin 118. Since today with a Ramps 1.4 Board (on a ANET A8)

Problem: Z_DUAL_STEPPER_DRIVERS

Now, my Z1 use the right direction, but my Z2 (on E1 slot). He rotate in the
wrong direction. I had try 
#define INVERT_E1_DIR true
but no changes.
how can i inverted the z2 stepper?

conf.h
https://pastebin.com/raw/2FtzfvYd

conf_adv.h
https://pastebin.com/raw/781n38tx

opy001, 2018-04-21:
I had a similar problem but mine was the pins_RAMPS.h file was not correct.  Mine did not have Z2 so I had to create my own.  In yours you may be able to just rename E1 to Z2.  Remember, this is a zero based system "most of the time" so E0 is extruder 1 and E1 is extruder 2.  This does not count for Z2.

Change this:
`#define E1_STEP_PIN        36`
`#define E1_DIR_PIN         34`
`#define E1_ENABLE_PIN      30`
`#define E1_CS_PIN          44`

To this:
`#define Z2_STEP_PIN        36`
`#define Z2_DIR_PIN         34`
`#define Z2_ENABLE_PIN      30`
`#define Z2_CS_PIN          44`

thinkyhead, 2018-04-22:
@noxx2015 —

- You don't need to add any Z2 pins to the RAMPS file. The pins for Z2 will be auto-assigned to the first free extruder plug automatically (in `pins.h`) in response to `Z_DUAL_STEPPER_DRIVERS`.
- At this time the direction bits for Z and Z2 have to match, so if you find that E1 goes the wrong direction, you should just flip the plug (or if the plug is keyed, reverse the pins inside the plug) and this will make Z2 move the same direction as Z.
- If this doesn't fully resolve the issue, then we recommend you do any further testing with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to ensure you're using the most up to date and bug-free Marlin.

As3uS, 2018-06-15:
Hi @thinkyhead is it necessary to activate dual enstop for Dual Z ? Or can i use only my Z probe as usual ?

thinkyhead, 2018-06-15:
> Hi @thinkyhead is it necessary to activate dual enstop for Dual Z?

No.

> Or can i use only my Z probe as usual ?

Yes.

As3uS, 2018-06-16:
Thank work fine ;)

closed 2018-07-01 by thinkyhead

o-nix, 2020-03-13:
For me, it was easier to reverse the plug. Thanks for the explanation.