MarlinFirmware/Marlin #10480: Anet a8 with GT2560 and BLTouch. Firmware wont upload.

jayenvy, 2018-04-22:
I am currently using the Anet A8 with GT2560 and BLTouch. I was able to get the firmware running on prusa i3 setting but has been a headache changing the settings for the Anet. I wanted to use the Anet A8 settings but the firmware is only recognized as anet v1. board and wont let me burn the Anet settings to the GT2560. When i flashed it to my Sanguino melzi 2,0 v5 board the firmware flashed fine but not with the GT2560. My main problem is the  Z offset, because the z offset is -0.10 i can only pick a first layer of 0.015 with a 0.1 layer height. it will not allow me to do a 0.2 height with the brim printed. The prim falls under the build window and forces me to do 0.1 which is a long print time as well as I cant get the nozzle correct for first layer. So I just want to get the Anet settings so this process will be alot easier. Is there something more I have to do? Ive searched youtube and google with no corrections to this issue. I think I have the RC7 firmware version because it was a headache getting the BLTouch to work in the first place. I can supply my config file as well as my slicer and gcode settings.

jayenvy, 2018-04-22:
Can you please clarify by what you mean by talk to GT2560?

jayenvy, 2018-04-22:
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

jayenvy, 2018-04-22:
TRIED BOTH WITH SKYNET AND WITH MARLIN BOTH THE SAME THING WONT UPLOAD

jayenvy, 2018-04-22:
Thanks but my issue is not the 1284 board my issue is using the gt2560 with activated bltouch on the correct firmware settings. Whenever i try to upload it whether skynet 3d mod or marlin it does not recognize the board as a proper board for the settings or turn on the bltouch ive spent hours upon hours trying to find a config h file that works . i can upload under prusa and gt2560. But the options are always different from anet a8. I lost for thought unless someone already has a settings config h i can twerk to my anet

jayenvy, 2018-04-22:
And just so your option was not wasted i tried that as well with the same issue as follows. Exit status 1 #error "oops! Make sure you have "anet v1.0", "anet v1.0 (optiboot)" or "sanguino" selected from the "tools -> boards menu"

jayenvy, 2018-04-22:
Or if i can get proper bltouch settings for gt2560. That will work too with the gt2560 settings config h file. Ill just change the stepper and build bed options manually. But bltouch doesn not activate under that board. Ive defined the bltouch by removing //. Is there something else i have to remove or change for the board to recognize the bltouch? It turns on but does not respond to code or deploy or sepf test.

ManuelMcLure, 2018-04-22:
What I did when I switched from the Anet A8 board to RAMPS was grab the Anet A8 configuration and just change the `MOTHERBOARD` setting, leaving everything else the same. Something like that might work with the GT2560 since it's sort of like an Arduino Mega 2560 + RAMPS 1.4 on a single board.

I've seen several references on Google that say that the GT2560 doesn't come with a bootloader. If that is the case, you can't update it from Arduino without first burning a bootloader for which you'll need an ISP programmer.

This may be something that's better discussed on the RepRap Forums Controller subforum: [http://forums.reprap.org/list.php?13](http://forums.reprap.org/list.php?13)

jayenvy, 2018-04-22:
Im willing to give it a try can you upload your config h and adv h? Either on here or by email jayenvy@live.com

jayenvy, 2018-04-22:
Just switching the mother board name does not work with my gt2560. It says to plug in anet v1.

jayenvy, 2018-04-22:
I just need a working firmware for my setup. Ive been trying all day with no avail. Again i have a anet a8 came with a 1284p sanguino melzi 2.0 v5. I upgraded to gt2560 with bltouch. There is no working firmware for my setup can someone please help?

ManuelMcLure, 2018-04-22:
Here's my setup - I don't have a BLTouch but instead an inductive sensor. I have the Arduino environment set to "Arduino/Genuino Mega or Mega 2560".

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1936183/Configuration.zip)

But if there's no bootloader on your GT2560 this isn't going to help. And installing a bootloader is outside the purview of Marlin.

jayenvy, 2018-04-22:
That setup is for ramps board. There is different pin assignments for gt2560. But i did as well go ahead and try and i got a fan pin error which confirmed the different pin layouts. I need a anet a8 setting that will let me flash to a gt2560.

jayenvy, 2018-04-22:
And just to make it clear. I can flash firmware to my gt2560. The issue is not being able or having knowledge on how to flash. The issue is i cant use my printer setti gs on the gt2560 with bltouch activated. I can upload the gt2560 firmware but the setting are all wrong with different build platform settings then the anet a8. Ive even copied each different line to see if u could just transfer over my settings and nope gives a total different read out. My y n x is all backwards. My homing is not at correct values. So i just want a working anet a8 setting for the gt2560.

ManuelMcLure, 2018-04-22:
The only thing you should need to do to get the right pinouts is to change the `MOTHERBOARD` setting in Configuration.h to either `BOARD_GT2560_REV_A` or `BOARD_GT2560_REV_A_PLUS`.

ManuelMcLure, 2018-04-22:
I don't have a GT2560 so I can't test it on my printer, but I took my configuration file, changed the `MOTHERBOARD` from `MOTHERBOARD BOARD_RAMPS_14_EFB` to `BOARD_GT2560_REV_A` and compiled. Everything compiled fine except that I had to comment out the setting for `ENDSTOP_INTERRUPTS_FEATURES`. Once I did that Marlin compiled fine.

 If you're getting that "Oops!  Make sure you have 'Anet V1.0'..." error it's because you have `MOTHERBOARD` still set to `BOARD_ANET_10` somewhere - that's the only way you can get that error message. Check to ensure that it isn't being set in more than one place in `Configuration.h`.

jayenvy, 2018-04-22:
Ok i will try again let you know the results

jayenvy, 2018-04-23:
So far this is my error. I downloaded newest marlin. Downloaded newest arduino. Moved anet a8 to main folder. Changed board and the error is x min pin is not interrupt capable

ManuelMcLure, 2018-04-23:
As I mentioned above, you need to comment out the `ENDSTOP_INTERRUPTS_FEATURE` setting.

ManuelMcLure, 2018-04-23:
Also, the configuration file I provided is for the bugfix-1.1.x branch, not for 1.1.8.

jayenvy, 2018-04-23:
Fixed the interrupt but now i got start adc keypad pin not declared. I dont even know what that is.

jayenvy, 2018-04-23:
Nevermind i see that its a lcd option. Lms if i can change it and compile successfully

jayenvy, 2018-04-23:
Looks like my issue was a old version of arduino. I was using the one that i had received with skynet smh. Now can compile. Thanks bro. do you know how to get the bltouch recognized? I know i remove // from the begining but then what is that it?

thinkyhead, 2018-04-23:
That's it!
```
#define BLTOUCH
```
If the BLTouch is connected to Z Min then use `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`, otherwise use `Z_MIN_PROBE_ENDSTOP`.

jayenvy, 2018-04-23:
Yea i already have that active. Now what?

jayenvy, 2018-04-23:
Is that it i can upload now?

thinkyhead, 2018-04-23:
_That's it!_

jayenvy, 2018-04-23:
Great to hear thanx everyone 😀😀😀😀😀

closed 2018-04-23 by jayenvy