MarlinFirmware/Marlin #10488: Unable to compile CR-10 exmaple comfiguration

gronat, 2018-04-22:
### Description
Unable to compile using the example configuration files for CR-10.

### Steps to Reproduce
1. Clone Marlin firmware 1.1.8
2. Replace Configuration.h and Configuration_adv.h with corresponding files from example_configurations/Creality/CR-10/ folder.
3. Open Arduino IDE 1.8.5 (using Linux mint), double check that config files correspond to those in examples configurations.
4. Hit the compile button

**Expected behavior:**
Successful compilation
**Actual behavior:** [What actually happens]
Throws an error:

In file included from sketch/MarlinConfig.h:26:0,
                 from sketch/ultralcd.cpp:23:
pins_MELZI_CREALITY.h:54: error: pasting "/* st9720 CLK*/" and "_RPORT" does not give a valid preprocessing token
 #define LCD_PINS_D4     30 // st9720 CLK
                            ^
sketch/fastio.h:90:41: note: in definition of macro '_WRITE'
 #define _WRITE(IO, v) do { if (&(DIO ## IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                         ^
sketch/ultralcd_st7920_u8glib_rrd.h:82:3: note: in expansion of macro 'WRITE'
   WRITE(ST7920_CLK_PIN, LOW);        ST7920_DELAY_1; \
   ^
sketch/ultralcd_st7920_u8glib_rrd.h:28:25: note: in expansion of macro 'LCD_PINS_D4'
 #define ST7920_CLK_PIN  LCD_PINS_D4
.....

#### Additional Information
The same behavior observed using Arduino IDE 1.8.4

cjsoong, 2018-04-22:
@gronat   Download bugfix 1.1.x

gronat, 2018-04-22:
Just did, and used example_configurations/Creality/CR-10mini/ config files. Getting the same error.

cjsoong, 2018-04-22:
@gronat I compiled no error.
Maybe reinstall arduino ide.
[image]

gronat, 2018-04-22:
Just double checked the behavior on Linux Mint 18 and a fresh Arduino IDE 1.8.4 and 1.8.5, still the same. I tried the same on Win7 and fresh Arduino IDE 1.8.5, Sanguino board, Atmega 1284P 16MHz, Olikarus u8glib v1.19.1 and the Marlin 1.1.8 - bugfix 1.1.x branch, CR-10mini config files. The same error appear.
[image]

gronat, 2018-04-22:
OK, it can not be compiled with U8glib 1.19. It does work with U8glib 1.18.0 Can be marked as resolved.

Please, could you add this to the documentation as a note?

thinkyhead, 2018-04-22:
It's a bug in the compiler. To work around it, remove the end-of-line comment from the pins it complains about.

boelle, 2019-02-20:
@gronat still having issues?

DevilByMoonlight, 2019-02-22:
Hi, I'm sorry to say that I too am getting out of memory  errors compiling the stock CR10 example files in the 1.19 as well as the latest bugfix, I have tried this on multiple machines using both 1.8.5  and the 1.8.8 Arduino IDE, the error seems to indicate I have run out variable heap space but oddly enough the CR10S example compiles  perfectly fine, as well as the marlin configs for my other printers (each printer has its own separate marlin directory that is  archived in its own separate .zip file) my situation is am replacing the melzi  board for  a well tested ramps 1.4/mega combo in my cr10 ,  so I changed the board type to 43 in configuration.h and then changed the board  & processor type under the IDE's tools menu then hit verify - after a few moments I am greeted with this out of memory error -

snip----
Sketch uses 128390 bytes (50%) of program storage space. Maximum is 253952 bytes.
Global variables use 15113 bytes (184%) of dynamic memory, leaving -6921 bytes for local variables. Maximum is 8192 bytes.
----snip
hmm I thought the melzi had a lot less resources than the mega :/ 

in addition to reinstalling the IDE  on several different machines both M$ & linux in flavour I have also tried other  libraries  as well as reverting back to the u8glib 1.18.0 but alas no fix for me - the weird thing is the CR10S config files  compile just fine - so me thinks I will either have to use the CR10s firmware and try my best to trim the fat off of it or go from a blank template and manually punch in the settings  from the cr10's config files 

I am but a mere novice with these printers but am willing to have a go, if I have overlooked something I would very thankful to be pointed  in the right direction - thanks for your time and trouble

Darius051, 2019-02-25:
Hi Devil..
I have the same error. I too have upgraded from a CR-10 board to ramps1.6+mega 2560 and don't know what to do. The other strange thing is when I want to upload the sketch to the board it gives me a error: Error Oops! Make sure you have Sanguino selected from the tools menu. But I have selected mega 2560 because that is the board that I'm using. 
I must admit that I have installed a sanguino library and used a arduino uno to burn a bootloader on the cr-10 board but it is not connected now. If someone have an idea what to do I would be thankful. :-)

boelle, 2019-02-25:
maybe attach 2 config files as a zip file?
tried bugfix 2.0?

what version of arduino ide?
tried vs code and platform io ?

DevilByMoonlight, 2019-02-25:
> Hi Devil..
> I have the same error. I too have upgraded from a CR-10 board to ramps1.6+mega 2560 and don't know what to do. The other strange thing is when I want to upload the sketch to the board it [...]

I found the CR10s config files compile and work 

> Hi Devil..
> I have the same error. I too have upgraded from a CR-10 board to ramps1.6+mega 2560 and don't know what to do. The other strange thing is when I want to upload the sketch to the board it [...]

Hi Darius, I'm going to try using the CR10S  config files instead of the CR10 ones which are also in the example folder, as they compile just fine on the ramps/mega combo I'm using as a replacement for the creality melzi,

DevilByMoonlight, 2019-02-25:
> maybe attach 2 config files as a zip file?
> tried bugfix 2.0?
> 
> what version of arduino ide?
> tried vs code and platform io ?

Thanks for the reply, Ive tried  the arduino IDE versions 1.8.5 & 1.8.8  on a few different machines just incase an update has broken something... the files that are giving a bit of a headache are the stock cr10  example ones as found in the 1.19 & bugfix releases the only thing I changed was the board type  from the creative melzi to 43, in order to reflect the replacement board ( ramps/mega combo) then setting the cpu and board type's  to mega 2560 in the tools menu  and hitting  the 'tick' to compile, I have not tried the 2.0x branch as its still Beta and as the board is an 8 bit ramps1.4 & mega one anyway I thought I would stick with the 1.1.9,  I cant say I've tried platform IO as yet  and as such am totally unfamiliar with it, the issue I was having is I can change the board type of other configuration.h files and they compile just fine, I have done this a few times previously for when swapping different boards in an A8, Kossel linear plus & ender 3  but for some reason the stock CR10 one just doesnt want to work for me, I do realise its most likely something pretty simple that I am overlooking, but thanks again for trying to help me out...

Darius051, 2019-02-26:
> Thanks for the reply, Ive tried the arduino IDE versions 1.8.5 & 1.8.8 on a few different machines just incase an update has broken something... the files that are giving a bit of a headache are the [...]

Thanks for your informations. I have actually solved that situation by upgrading to IDE 1.8.8 and manually edit some lines only in the stock CR10 example config.h file and then replacing it with the original one of the bugfix 1.1.x release. I did not replace the _adv.h like before when the problem occurred so maybe there was something in there that was not right. And the sanguino problem that I mentioned did not appear again. Maybe it was because of the new 1.8.8 installation. Good luck :-)

thinkyhead, 2019-03-03:
The original CR-10 board has 16K of SRAM. The CR-10S board has only 8K. So use the CR-10S configuration files as guide for good CR-10S settings, not the CR-10 configuration files.

closed 2019-03-03 by thinkyhead

DevilByMoonlight, 2019-03-05:
Thanks I'm  now currently using  a slightly tweaked  CR10S's Marlin  bugfix 1.1.x configuration on my modded  mega clone & RAMPS 1.4 that I've managed to shoehorn into the CR10's control box - am running LV8729's  with good success - and am very happy with it - at 32 steps its not as quiet as the 2130's on my kossel, but much better than stock - once its done a few more hours printing will sort out the box's internal cable management  to make sure  nothing interrupts the air flow  and lock it up,