MarlinFirmware/Marlin #10489: [bugfix-2.0.x]  Error compiling ArduinoIDE

mtmello2003, 2018-04-22:
I´ve got this error trying to compile Marlin 2.0:

Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: file name or extension too big.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

teemuatlut, 2018-04-22:
Use IDE version 1.9.0 or modify your `build.path`.

p3p, 2018-04-22:
You need to use Arduino 1.9 beta or PlatformIO to compile Marlin 2.0, there is a bug in the Arduino 1.8.

@thinkyhead I think we need to clarify this somewhere easy to notice as more people are realising Marlin 2.0 beta exists

mtmello2003, 2018-04-22:
Tks... I´m downloading 1.9 beta and... A lot of people I know, think that Marlin 2 run only in 32bits hardware...

Ludy87, 2018-04-22:
Referce to https://github.com/MarlinFirmware/Marlin/issues/10447

mtmello2003, 2018-04-23:
Yes, I got success with IDE 1.9.0 Beta.

Thanks.

closed 2018-04-30 by thinkyhead