MarlinFirmware/Marlin #10492: {bugfix-2.0.x] Not reverting stepper motor

mtmello2003, 2018-04-22:
Using:
// A dual extruder that uses a single stepper motor
#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
#define SWITCHING_EXTRUDER_SERVO_NR 1
#define SWITCHING_EXTRUDER_SERVO_ANGLES { 145, 25 } // Angles for E0, E1[, E2, E3]
#if EXTRUDERS > 3
#define SWITCHING_EXTRUDER_E23_SERVO_NR 1
#endif
#endif

When I use the script to change between extruders:
  #define USER_DESC_4 "Change to E2"
  #define USER_GCODE_4 "M203 E400\nG92 E0\nG91\nG1 E-3 F120\nG1 E-2.5 F60\nG1 E-120.5 F18000\nG92 E0\nG91\nT1\nG1 E+120.5 F18000\nG1 E+2.5 F60\nG1 E+3 F120\nM203 E25\nG92 E0\nG90"
  
  #define USER_DESC_5 "Change to E1"
  #define USER_GCODE_5 "M203 E400\nG92 E0\nG91\nG1 E-3 F120\nG1 E-2.5 F60\nG1 E-120.5 F18000\nG92 E0\nG91\nT0\nG1 E+120.5 F18000\nG1 E+2.5 F60\nG1 E+3 F120\nM203 E25\nG92 E0\nG90"

The stepper motor do not auto reverse in E2.

This issue do not happen in 1.1.x.

Thanks

mtmello2003, 2018-04-22:
Hey, I got the trigger that start the issue: 

#define LIN_ADVANCE

If I disable the Linear advance, the issue do not occur and the stepper motor revert happen with success.

mtmello2003, 2018-04-30:
I think I didn´t get the point you asking but if it helps: when I run the T1 gcode, the stepper motor is reverted and return back rotation with T0.

Sebastianv650, 2018-05-02:
OK, the problem is (again, like for duplication mode) in the stepper control macros, found in:
- stepper_indirection.h
- stepper.cpp

My problem is I always end up confused when looking at these macros using macros using.. It's like standing between two mirrors, looking into infinity X-)
Try the follwing, in stepper.cpp, inside `void Stepper::advance_isr()` you will find an #if #endif block like this:

```cpp
    #if ENABLED(MK2_MULTIPLEXER) // For SNMM even-numbered steppers are reversed
      #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) E## INDEX ##_DIR_WRITE(e_steps < 0 ? !INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0) : INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0)); }while(0)
    #elif ENABLED(DUAL_X_CARRIAGE) || ENABLED(DUAL_NOZZLE_DUPLICATION_MODE)
      #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)
    #else
      #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) E## INDEX ##_DIR_WRITE(e_steps < 0 ? INVERT_E## INDEX ##_DIR : !INVERT_E## INDEX ##_DIR); }while(0)
    #endif
```

Add a line (note the SWITCHING_EXTRUDER one) so it becomes:

```cpp
    #if ENABLED(MK2_MULTIPLEXER) // For SNMM even-numbered steppers are reversed
      #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) E## INDEX ##_DIR_WRITE(e_steps < 0 ? !INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0) : INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0)); }while(0)
    #elif ENABLED(DUAL_X_CARRIAGE) || ENABLED(DUAL_NOZZLE_DUPLICATION_MODE)
      #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)
    #elif ENABLED(SWITCHING_EXTRUDER)
      #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) E## INDEX ##_DIR_WRITE(e_steps < 0 ? INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0) : !INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0)); }while(0)
    #else
      #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) E## INDEX ##_DIR_WRITE(e_steps < 0 ? INVERT_E## INDEX ##_DIR : !INVERT_E## INDEX ##_DIR); }while(0)
    #endif
```

If it works, I have to check what has to be done to support more than T0 and T1.

mtmello2003, 2018-05-02:
I did the code changes in stepper.cpp but unfortunately didn´t work.

Same issue: with LA activated the extruder motor do not reverse. When I comment the tag //#define LIN_ADVANCE, works okay.

The code altered seems do not take effect in LA behavior.

Sebastianv650, 2018-05-03:
That's realy strange as I just adopted the MK2_MULTIPLEXER define which I assume is working. @thinkyhead, any idea? Every second extruder needs direction reversement, so the only affected macro should be SET_E_STEP_DIR, which I tried?

thinkyhead, 2018-05-08:
Try this instead:
```cpp
void Stepper::advance_isr() {

  #if ENABLED(MK2_MULTIPLEXER) // For SNMM even-numbered steppers are reversed
    #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) E0_DIR_WRITE(e_steps < 0 ? !INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0) : INVERT_E## INDEX ##_DIR ^ TEST(INDEX, 0)); }while(0)
  #elif ENABLED(DUAL_X_CARRIAGE) || ENABLED(DUAL_NOZZLE_DUPLICATION_MODE)
    #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)
  #elif ENABLED(SWITCHING_EXTRUDER)
    #if EXTRUDERS == 2
      #define SET_E_STEP_DIR(INDEX) do { if (e_steps < 0) \
             E0_DIR_WRITE(INDEX ? !INVERT_E0_DIR : INVERT_E0_DIR); \
        else E0_DIR_WRITE(INDEX ? INVERT_E0_DIR : !INVERT_E0_DIR); \
      }while(0)
    #elif EXTRUDERS > 4
      #define SET_E_STEP_DIR(INDEX) do { if (e_steps < 0) \
             switch (INDEX) { case 0: E0_DIR_WRITE( INVERT_E0_DIR); break; case 1: E0_DIR_WRITE(!INVERT_E0_DIR); break; case 2: E1_DIR_WRITE( INVERT_E1_DIR); break; case 3: E1_DIR_WRITE(!INVERT_E1_DIR); break; case 4: E2_DIR_WRITE( INVERT_E2_DIR); } \
        else switch (INDEX) { case 0: E0_DIR_WRITE(!INVERT_E0_DIR); break; case 1: E0_DIR_WRITE( INVERT_E0_DIR); break; case 2: E1_DIR_WRITE(!INVERT_E1_DIR); break; case 3: E1_DIR_WRITE( INVERT_E1_DIR); break; case 4: E2_DIR_WRITE(!INVERT_E2_DIR); } \
      }while(0)
    #elif EXTRUDERS > 2
      #define SET_E_STEP_DIR(INDEX) do { if (e_steps < 0) \
             switch (INDEX) { case 0: E0_DIR_WRITE( INVERT_E0_DIR); break; case 1: E0_DIR_WRITE(!INVERT_E0_DIR); break; case 2: E1_DIR_WRITE( INVERT_E1_DIR); break; case 3: E1_DIR_WRITE(!INVERT_E1_DIR); } \
        else switch (INDEX) { case 0: E0_DIR_WRITE(!INVERT_E0_DIR); break; case 1: E0_DIR_WRITE( INVERT_E0_DIR); break; case 2: E1_DIR_WRITE(!INVERT_E1_DIR); break; case 3: E1_DIR_WRITE( INVERT_E1_DIR); } \
      }while(0)
    #endif
  #else
    #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) E## INDEX ##_DIR_WRITE(e_steps < 0 ? INVERT_E## INDEX ##_DIR : !INVERT_E## INDEX ##_DIR); }while(0)
  #endif

  #if ENABLED(DUAL_X_CARRIAGE) || ENABLED(DUAL_NOZZLE_DUPLICATION_MODE)
    #define START_E_PULSE(INDEX) do{ if (e_steps) E_STEP_WRITE(!INVERT_E_STEP_PIN); }while(0)
    #define STOP_E_PULSE(INDEX) do{ if (e_steps) { E_STEP_WRITE(INVERT_E_STEP_PIN); e_steps < 0 ? ++e_steps : --e_steps; } }while(0)
  #else
    #define START_E_PULSE(INDEX) do{ if (e_steps) E## INDEX ##_STEP_WRITE(!INVERT_E_STEP_PIN); }while(0)
    #define STOP_E_PULSE(INDEX) do { if (e_steps) { e_steps < 0 ? ++e_steps : --e_steps; E## INDEX ##_STEP_WRITE(INVERT_E_STEP_PIN); } }while(0)
  #endif

  . . .
```

thinkyhead, 2018-05-19:
Additional patches have been made to the bugfix branches, as both the direction and stepping behavior needed fixing, so please test again.

Your compile errors are due to not having `E2_DIR_PIN` defined.

mtmello2003, 2018-05-20:
I´m still having error to compile with the LINEAR_ADVANCE uncommented:

first lines:

In file included from sketch\src\module\../inc/../HAL/HAL_AVR/HAL.h:47:0,

                 from sketch\src\module\../inc/MarlinConfig.h:28,

                 from sketch\src\module\stepper_indirection.h:47,

                 from sketch\src\module\stepper.h:46,

                 from sketch\src\module\stepper.cpp:57:

sketch\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:79:41: error: pasting "DIO" and "-" does not give a valid preprocessing token

 #define _WRITE(IO,V)          do{ if (&(DIO ## IO ## _RPORT) < (uint8_t*)0x100) _WRITE_NC(IO,V); else _WRITE_C(IO,V); }while(0)

                                         ^

I downloaded the latest version of 2.0 bugfix

closed 2018-05-23 by thinkyhead