MarlinFirmware/Marlin #10493: [1.1.x - Bugfix] Auto bed leveling goes below z offset.

forkoz, 2018-04-23:
### Description

### Steps to Reproduce
Find Z offset
M851 Z X.XX
M500
M501
G29 (bilinear)
M500
**Expected behavior:** [What you expect to happen]
Auto bed bed leveling works with offsets like -.01, -.03, etc

**Actual behavior:** [What actually happens]
Z=-1.5
Receive -1.57, -1.60, -1.4
Nozzle Grinds bed.

#### Additional Information

I tore my hair out and wasted some tape but realized ABL goes from wherever you leave the head. Maybe this needs a warning or some catch. Like make it force a home of Z first.

thinkyhead, 2018-04-23:
Bed leveling does require your Z=0 to be set correctly, but it shouldn't matter at all if you move the nozzle around between `G28` and `G29`. If you let the steppers go to sleep then `G29` does in fact require a re-homing.

If you're getting weird readings from bed leveling it's more likely that your `M851 Z` offset is inaccurate. But if you want to do more testing and give us a log, then we can see exactly what's happening. Here's the boilerplate instructions:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do whatever it is that you think is messing with your leveling.
- Do a `G29` to probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

(Note that `M501` after `M500` is not needed, not even with UBL which formerly required it.)

FiCacador, 2018-04-25:
I too was having this problem with bugfix-1.1.x, using bilinear on a Creality with a Melzi board.

The problem was that the probe triggered at a certain nozzle height when doing G28, from where I get my z offset, but then triggered at a lower nozzle height when doing G29, a difference of about 1mm, which is huge, and resulted in having the nozzle squeezed  onto the bed.

Off course there is an easy workaround which is to measure z-offset after G29, but that's not exactly the ideal solution.

What I observed was that there is a reaction time for the board to detect when the probe triggers, and that was the root of this problem, for two reasons:

- The probe moves slower when doing G28 than when doing G29:
#define HOMING_BUMP_DIVISOR { 2, 2, 4 }  // Re-Bump Speed Divisor (Divides the Homing Feedrate)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
- The reaction time for detecting that the probe has triggered is shorter on G28 than on G29.

My solution was to make the probe speeds the same on both situations, lowering the G29 probing speed:
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 4)

That made thing better, but wasn't enough, so I tried disabling segment leveled moves:
//#define SEGMENT_LEVELED_MOVES
//#define LEVELED_SEGMENT_LENGTH 5.0 // (mm) Length of all segments (except the last one)

Again things got better, but there was still a slight difference, so I also disabled leveling fade height:
//#define ENABLE_LEVELING_FADE_HEIGHT

And that was it, now z-offset auto bed level is working as well as it was before on a previous version that didn't had any of this features that I disabled.

Was it too much to the Melzi board to handle?
Anyway, it makes no sense to be Z homing and probing at different speeds.

forkoz, 2018-04-25:
Yes it did it again, at Z.40 it "corrects" for Z 1.20? SO either I comment those out or home and level first at Z=0?

[Log-z.40.txt](https://github.com/MarlinFirmware/Marlin/files/1947064/Log-z.40.txt)
[Log-Z0.txt](https://github.com/MarlinFirmware/Marlin/files/1947073/Log-Z0.txt)

Looxoor, 2018-04-26:
I just wanted to chime in that it happened to me last night as well, for the first time.
I'm on 1.1.8.

I had just relocated my probe to a new position, then did the following steps:

- In configuration.h I added the new probe-to-nozzle offsets, changed my Z offset to 0, reflashed.
- Heated bed and nozzle to 60/215, performed a Z offset calibration, added new value to configuration.h and reflashed.
- Heated bed and nozzle again, entered G28 followed by G29.
On the eighth point out of nine the nozzle went straight into the bed before the probe triggered and it began to raise to do the slow probing when I cut the power.

Come to think about it, I had the old Z offset value stored in eeprom, do I have to redefine it with M851 M500 as well as in the firmware? If so then I know what the problem is as the old value was about a mm lower than the new one.

forkoz, 2018-04-27:
Yes, I learned when I changed to 1/32 stepping, old values were still in ee when I reflashed the fw and I reflashed a couple of times wondering what happened.

I've been working around this by homing first after setting z. Also tried UBL and its a slight improvement. Problem is with my new cooler my probe offset on X is 48 and this causes issues as marlin centers on the probe leaving the nozzle quite left of center. I had luck manually jogging the nozzle to X=100 and taking the initial z offset there. Then I run UBL as normal. More points for bilinear and less for UBL also help. 3 is too few on 200mm bed, 10 is too many.

Looxoor, 2018-04-27:
Retried yesterday, first by lowering the probe so it triggers earlier, readjusting the min Z value, save to eeprom (and config) and presto!

Turns out the bed was so skewed on a certain point that the nozzle could catch before the probe could trigger at it's position of 60 mm offset in the y axis.

So, be sure to have as early trigger as feasible.

forkoz, 2018-04-27:
That part on mine was fine. It was just recording values that fell below the Z offset, probably because it started recording them at the Z offset and not 0. I'm working on getting the probe closer to the nozzle and using UBL since I still have 30% free with it and debugging compiled in. I don't know if UBL has the issue because I'm homing after setting Z now always.

thinkyhead, 2018-04-28:
Here's a change that seems like a good idea:
```diff
inline float get_homing_bump_feedrate(const AxisEnum axis) {
+ #if HOMING_Z_WITH_PROBE
+   if (axis == Z_AXIS) return Z_PROBE_SPEED_SLOW;
+ #endif
  static const uint8_t homing_bump_divisor[] PROGMEM = HOMING_BUMP_DIVISOR;
  uint8_t hbd = pgm_read_byte(&homing_bump_divisor[axis]);
  if (hbd < 1) {
    hbd = 10;
    SERIAL_ECHO_START();
    SERIAL_ECHOLNPGM("Warning: Homing Bump Divisor < 1");
  }
  return homing_feedrate(axis) / hbd;
}
```

thinkyhead, 2018-04-28:
> Come to think about it, I had the old Z offset value stored in eeprom, do I have to redefine it with M851 M500 as well as in the firmware?

Oh yes. Or, `M502`, `M500` to use the value in the configuration.

boelle, 2019-02-20:
@forkoz still having an issue if using the latest bugfix 2.0?

closed 2019-07-11 by shitcreek