MarlinFirmware/Marlin #10497: [2.0.x] TMC2130　Issue of sensorless homing at Delta

JXPA, 2018-04-23:
Hello!
I tried sensorless homing of TMC2130.
StallGuard2 works correctly and works wonderfully.
However, there is a problem with the behavior during homing.

If sensorless homing is disabled, all axes start to rise same time when homing starts.
However, if sensorless homing is enabled, only the X tower will move first.
The Y tower moves after the X tower hit, and the Z tower moves after the Y tower hit.
So, when the nozzle is in a low position, the Effector collides with the tower while the carriage moves.
Therefore, G33 can not be executed and auto calibration can not be done.

Is there a setting to move all the towers same time with sensorless homing enabled?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1937271/Configuration.zip)

JXPA, 2018-04-24:
Thank you.
Sensorless homing is attractive, but I will put it back in the optical endstop.

closed 2018-04-24 by JXPA

JXPA, 2018-04-26:
I tried changing some conditions.
By setting the acceleration and jerk to be smaller, we can now find the setting to start moving at the same time.
The threshold is very delicate, one less reacts excessively, one more does not react even if the carriage hits.

I hear a loud sound different from the collision sound immediately after the carriage hits.
It seems to be the sound when the stepper motor became active.
When "G33 P1" is executed and the height is measured, a value 10 mm larger than the actual height is displayed.

I think it is very difficult to adjust for using this function in my delta.

Thank you.

thinkyhead, 2018-04-26:
> The threshold is very delicate, one less reacts excessively, one more does not react even if the carriage hits.

@JXPA — You're not alone. The threshold can be very fiddly.

JXPA, 2018-04-26:
Thank you.
The threshold was somewhat widened by changing belt tension, microstep, motor voltage etc as well.
However, just after the carriage hit, the sound "BANG! BANG! BANG!" Will be heard from each axis.

I looked into the timing when the sound occurred.
For that, I have set HOME_BUMP_MM to 50 and HOMING_BUMP_DIVISOR to {1, 1, 1}.

The X carriage to a position higher than the Y, Z carriage and then execute G 28.
Observation result:
1. Carriages move up.
2. The X carriage hits.
3. The other two carriages are stopped.
4. I hear the sound that occurs when the motor becomes active from each axis (BANG! BANG! BANG!)(Belt also vibrates greatly)
5. Re-bump the X carriage to 50 mm.
6. Re-bump the Y and Z in the same way.
7. Move down to safe zone.

I could not erase this sound no matter what setting I made in the experiment.

thinkyhead, 2018-04-26:
> `HOME_BUMP_MM` to 50

That's rather excessive. A value under 10 should be sufficient.

thinkyhead, 2018-04-26:
> I could not erase this sound no matter what setting I made in the experiment.

I cannot account for this. Maybe it's something peculiar to turning stealthchop on and off.