MarlinFirmware/Marlin #10504: [1.1.8] Can not verify sketch

HansGrub3r, 2018-04-24:
Just configured Marlin for my mini kossel the way I want it. But when I felt done and pressed verify, it seems to be done but throws a error on me.
Without any real explanation on what is wrong, at first I thuoght it was my old copy of Arduino IDE 1.6.12 which were acting up on me since it's 2018 already, uninstalled it and got the latest 1.8.5 but still unable to verify the sketch.

Anyone able to shed some light on whats wrong with my attempt to verifying ?
It just says "open NUL: The system cannot find the file specified.
Error compiling for board Arduino/Genuino Mega or Mega 2560."

Tried google and not found anything close to this nor a solution to it.
At first it said this line of text aswell:
"This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences."

And after doing so it spit out this, which still wont let me know whats wrong.
I've got the right com port selected aswell as bord and processor.

> Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Jimmy\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751 -warnings=none -build-cache C:\Users\Jimmy\AppData\Local\Temp\arduino_cache_700229 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Jimmy\Downloads\Marlin-1.1.x (1)\Marlin-1.1.x\Marlin\Marlin.ino
> C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Jimmy\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751 -warnings=none -build-cache C:\Users\Jimmy\AppData\Local\Temp\arduino_cache_700229 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Jimmy\Downloads\Marlin-1.1.x (1)\Marlin-1.1.x\Marlin\Marlin.ino
> Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
> Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
> Detecting libraries used...
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\Marlin.ino.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\G26_Mesh_Validation_Tool.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\I2CPositionEncoder.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\M100_Free_Mem_Chk.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\MarlinSerial.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\Marlin_main.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\Max7219_Debug_LEDs.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\Sd2Card.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\SdBaseFile.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\SdFatUtil.cpp" -o "nul"
[... cut ...] 0L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\sketch\Marlin.ino.cpp" -o "C:\Users\Jimmy\AppData\Local\Temp\arduino_build_518751\preproc\ctags_target_for_gcc_minus_e.cpp"
> open NUL: The system cannot find the file specified.
> Error compiling for board Arduino/Genuino Mega or Mega 2560.

thinkyhead, 2018-04-24:
- Quit Arduino IDE.
- Rename, move, or delete your Arduino Sketchbook folder.
- Restart Arduino IDE.
- Close all windows.
- Open the `Marlin.ino` file in your project.
- Try Verify again.

boelle, 2019-02-20:
@HansGrub3r 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

HansGrub3r, 2019-02-20:
This was posted 10 months ago just to let you know.
Were going to say that this was quite a necro but apparently this was still open.

closed 2019-02-20 by HansGrub3r

MeJasonT, 2020-06-25:
@boelle looks like a bug to me.
either somthing in marlin is not compiling and the Arduino IDE is hanging
or what i think is the issue The arduino IDE on windows 10 is buggy
I came to this very justifable reason as sometimes the IDE wont open and i have to go delete a file 
C:\Users\MTEC\AppData\Local (Arduino 15 folder) this makes it work for a while.
As the Arduino IDE is the tool many of us use to laod marlin firmware then any bug in the tool chain is a bug affecting the use of Marlin. This is a current un fixed bug with the IDE.  Its possible this bug originates from windows anal security measures regarding file permissions.  The number of times i get told i dont have permission to save my own work to my own folders is now beyond a joke.  I am the administrator and still with various buggy updates my ability to use my pc without fault is now non existant.  Win 7 no issues another pc with an upgraded win7 to win10  has so many faults microsoft should usen it as a test bed - it might improve their shite.  The clean win 10 build is the one that wont verify, correction it will the example programs just not marlin.  Please dont fall off that high horse you are on but im affraid this is posted in exactly the correct group.

boelle, 2020-06-25:
@MeJasonT dont use arduino ide... use vscode+platform.io

we say this many times as the arduino ide is just outright horrible

thinkyhead, 2020-06-26:
@MeJasonT — Thank you for your insights.