MarlinFirmware/Marlin #10505: bugfix-1.1.x compile error: 'current_temperature_chamber' was not declared in this scope.

StevenBruinen, 2018-04-24:
### Description

Compile error with April 24th 1.1.x bugfix build:

temperature.cpp: In static member function 'static void Temperature::checkExtruderAutoFans()':

temperature.cpp:562: error: 'current_temperature_chamber' was not declared in this scope

     if (current_temperature_chamber > EXTRUDER_AUTO_FAN_TEMPERATURE)

### Steps to Reproduce

Compile the April 24thl bugfix-1.1.x 

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1942496/Configuration.zip)

StevenBruinen, 2018-04-24:
After looking in the code I see that "Temperature::current_temperature_chamber" only is available when HAS_TEMP_CHAMBER is enabled but the variable is also used in the HAS_AUTO_FAN code. 

#if HAS_TEMP_CHAMBER
  float Temperature::current_temperature_chamber = 0.0;
  int16_t Temperature::current_temperature_chamber_raw = 0;
  uint16_t Temperature::raw_temp_chamber_value = 0;
#endif

#if HAS_AUTO_FAN

[...]

  if (current_temperature_chamber > EXTRUDER_AUTO_FAN_TEMPERATURE)
    SBI(fanState, pgm_read_byte(&fanBit[5]));

systemik, 2018-04-24:
I do have the same issue and don't understand how to fix it. Function declaration is not done because this define is not trigered : #define HAS_TEMP_CHAMBER (PIN_EXISTS(TEMP_CHAMBER) && TEMP_SENSOR_CHAMBER != 0 && TEMP_SENSOR_CHAMBER > -2)

StevenBruinen, 2018-04-24:
For now I've disabled the code in "temperature.ccp" in the HAS_AUTO_FAN section by addding // in front of them.

#if HAS_AUTO_FAN

[...]

// if (current_temperature_chamber > EXTRUDER_AUTO_FAN_TEMPERATURE)
// SBI(fanState, pgm_read_byte(&fanBit[5]));

thinkyhead, 2018-04-24:
Thanks for the heads-up. Patched!

commit ad3b95c8 (2018-04-24): Fix current_temperature_chamber compile error

Fix #10505
commit 9789c1c7 (2018-04-24): Fix current_temperature_chamber compile error

Fix #10505
closed 2018-04-24 by thinkyhead