MarlinFirmware/Marlin #10506: [FEATURE] More thermal runaway configuration

Zer0t3ch, 2018-04-24:
It would be nice to see more settings for thermal runaway. Specifically, I'd like to see the ability to disable the hotbed but continue the print in the case of hotbed thermal runaway, but I'm sure there are other configurations that could be interesting/helpful.

thinkyhead, 2018-04-24:
It's an interesting concept. Note that if the bed heater turns off, there's a high probability that the print will come off the bed. The bed will also contract. And, there's a possibility that for certain types of error the bed will not really be turned off, but will continue heating out of control. So care must be taken.

Zer0t3ch, 2018-04-24:
> there's a high probability that the print will come off the bed

How would that be different from when the print is done, or just printing cold? I almost always have a difficult time getting my prints off the bed after it's cooled down, and I've had moderate success printing on a cold bed. (~30% fail in the first 3 or 4 layers, no problems after they pass that) 

> And, there's a possibility that for certain types of error the bed will not really be turned off, but will continue heating out of control. So care must be taken

Isn't there some kind of "min temp"? And if not, we could implement one. To clarify: if a heater wire comes loose and it can't heat, but the thermistor is still attached and drops slowly to room temp, there's clearly no significant danger, the print can continue. But, if the thermistor disconnects and suddenly reads 0, we can allow for a stop.

Which brings me to another point: it would be nice to implement some kind of pausing/nozzle parking in the case of certain types of thermal runaway while causing a hard shutdown on others. Like if my hotend thermistor breaks and it's reading 0: might as well hard stop. Just in case there's a short on the heating element, don't want it to somehow keep heating. Same for the bed, probably. But gradual cooldowns with a still-functional thermistor seems like the prime use-case for a print pause. 

Presumably some kind of indicator to whatever device is running the print (in the case of a non-SD card print) but I'm sure we could work with Octopi, Cura, and such to add handling for such a case.