MarlinFirmware/Marlin #10508: Manual step-by-step calibration

xero83, 2018-04-24:
Hi,
I plan to do the following for my machine: Make a type of calibration that goes through 4 points of the bed, and in each one of them it stops to raise or lower the bed, that is, on the screen, it puts calibration on, and stops at each point, correct one by one, and give the screen the next, I don't know if I know if I'm explaining it. It would be like witbox or ultimaker.
Marlin provides this option? And could I also keep what I want to do besides the autolevel when I start printing?

Thank you very much.

Translated with www.DeepL.com/Translator

boelle, 2018-04-24:
/**
 * The "Manual Probe" provides a means to do "Auto" Bed Leveling without a probe.
 * Use G29 repeatedly, adjusting the Z height at each point with movement commands
 * or (with LCD_BED_LEVELING) the LCD controller.
 */
//#define PROBE_MANUALLY

thinkyhead, 2018-04-26:
The feature you're looking for is this:
```cpp
// Add a menu item to move between bed corners for manual bed adjustment
//#define LEVEL_BED_CORNERS

#if ENABLED(LEVEL_BED_CORNERS)
  #define LEVEL_CORNERS_INSET 30    // (mm) An inset for corner leveling
  //#define LEVEL_CENTER_TOO        // Move to the center after the last corner
#endif
```

closed 2018-04-26 by thinkyhead