MarlinFirmware/Marlin #10512: [bugfix-2.0.x] compile error when CONTROLLER_FAN_PIN set

boelle, 2018-04-24:
i get this error when i try to compile: https://pastebin.com/eWw4Qsv3

its sure a fault of mine but i cant figure it

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1944413/Configuration.zip)

Roxy-3D, 2018-04-24:
#define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
 
This is almost for sure a result of this Pull Request:  https://github.com/MarlinFirmware/Marlin/pull/10502

This is an attempt to save a few clock cycles within the ISR's and is a good thing.   But apparently, it isn't fully baked yet.    If you back up and grab a copy of Marlin just prior to this,  or back this Pull Request out...    The problem should go away.     My guess is this will get fixed tonight.

p3p, 2018-04-24:
Looks like the Arduino IDE bug where its preprocessor gets comments inserted into token concatenations

ejtagle, 2018-04-24:
This is a avr-gcc bug, and is not being compliant with C99 
We can only workaround it:

Just replace
#define CONTROLLER_FAN_PIN 4 // Set a custom pin for the controller fan
With
#define CONTROLLER_FAN_PIN 4

And it will compile and work. That is the only workaround i found for this problem so far

boelle, 2018-04-24:
my "more support like q":

just as a convinence i have set it to home to X and Y max so it matches the mk42 heatbed (i dont use the spec 9 probe points)

but my endstops are connected to X min and Y min,

So i wonder if the most easy is to

#define USE_XMAX_PLUG
#define USE_YMAX_PLUG

and then in the ramps pin file flip the pin numbers ?

in repetier i could just use the online config tool and tell that the max endstops was connected to the min plug

we can always delte posts to keep it to topic but i'm just to damm curious to wait for me being added to the facebook page

boelle, 2018-04-24:
anyways errors are now down to:

C:\Users\boher\AppData\Local\Temp\arduino_build_442589\sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp:72:20: fatal error: U8glib.h: No such file or directory

 #include <U8glib.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

its years since i have used marlin, do i need to install this lib also?

ejtagle, 2018-04-25:
Yes, you need to install u8glib

closed 2018-04-26 by thinkyhead