MarlinFirmware/Marlin #10515: [bugfix-2.0.x] compile error 'class Temperature'

v-mat, 2018-04-25:
I'm getting the following error when I try to compile the latest bugfix-2.0.x:

"
sketch/src/gcode/config/M304.cpp: In static member function 'static void GcodeSuite::M304()':
sketch/src/gcode/config/M304.cpp:31:40: error: 'class Temperature' has no member named 'bedKp'
   if (parser.seen('P')) thermalManager.bedKp = parser.value_float();
                                        ^
sketch/src/gcode/config/M304.cpp:32:40: error: 'class Temperature' has no member named 'bedKi'
   if (parser.seen('I')) thermalManager.bedKi = scalePID_i(parser.value_float());
...
"

I'm able to compile (no errors) though if I comment #define PIDTEMPBED.

p3p, 2018-04-25:
Hey @v-mat, the PIDTEMPBED feature requires that you have a heated bed to control, you need to specify that you have one in the config by setting `TEMP_SENSOR_BED` appropriately.

v-mat, 2018-04-25:
Oops, I did miss that. Thanks!

closed 2018-04-25 by v-mat

thinkyhead, 2018-04-26:
Tweaked Marlin to make sure `PIDTEMPBED` is disabled if there's no bed heater.