MarlinFirmware/Marlin #10518: Digipots for sbase board current settings not taken into account

pgrandin, 2018-04-25:
### Description

```
#define DIGIPOT_I2C_MOTOR_CURRENTS { 0.1, 1, 0.7, 1, 0.7 }  //  MKS SBASE: 5
```
doesn't seem to be taken into account (and i do have `#define DIGIPOT_I2C` uncommented)

### Steps to Reproduce

1. Enable DIGIPOT_I2C (for use with a MKS SBASE V1.3)
2. Set current values for each of the 5 steppers
3. Compile and flash

**Expected behavior:** stepper current correctly managed .

**Actual behavior:** My main issue is that the extruder won't move. As a test I tried to reduce the current for the X-axis stepper (reflected in the attached config) but I didn't see any difference

The printer is working correctly using smoothieware, so I know that my wiring is correct, and I do have some decent reference values from the smoothieware config.

I am testing using commit a90cbc633993dc236fdb5d2aa174b52c321533c4 (HEAD -> bugfix-2.0.x, origin/bugfix-2.0.x) which has been reported to work by other users ( https://github.com/MarlinFirmware/Marlin/pull/9927 )

Thanks!
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1945737/marlin.zip)

SwiftingSpeed, 2018-04-25:
Have you heated the extruder nozzle? the stepper won't move if it's not up to temp.
Configuration.h:
```
// @section extruder

// This option prevents extrusion if the temperature is below EXTRUDE_MINTEMP.
// It also enables the M302 command to set the minimum extrusion temperature
// or to allow moving the extruder regardless of the hotend temperature.
// *** IT IS HIGHLY RECOMMENDED TO LEAVE THIS OPTION ENABLED! ***
#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 170
```
I can confirm that the extruder stepper driver is working on MKS Sbase, I have it working myself.

pgrandin, 2018-04-25:
:man_facepalming: 

Thanks @SwiftingSpeed

closed 2018-04-25 by pgrandin