MarlinFirmware/Marlin #10525: [Bugfix-1.1.x] Moving while z-probe is triggered, impacts delta kinematics.

joris57, 2018-04-25:
On a delta, after homing, a forward and backward movement of 10 mm is executed, while the inductive z-probe is triggered..

Expected: position in both mm and steps should be identical before and after.

Observed:
1) Position in steps is NOT identical (in mm: ok)
2) One of the movements is not fully performed (just a small step, not the full 10mm)
3) Kinematics afterward are affected. Instructions for movement in x and y, seem to result also in movement in z (effector moves on an arc, rather than in the plane.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1947762/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1947763/Configuration_adv.h.txt)

Gcode:
```
G28
Recv: X:0.00 Y:0.00 Z:266.84 E:0.00 Count X:46255 Y:46255 Z:46255
G91; G1 Y10 F6000; G90
G91; G1 Y-10 F6000; G90
Recv: echo:endstops hit:  Z:467.22
M114
Recv: X:0.00 Y:0.00 Z:266.84 E:0.00 Count X:45984 Y:45984 Z:46720
```

thinkyhead, 2018-04-26:
What is your use-case for `G30`? It's generally meant for troubleshooting or finding a probe-to-bed distance in a one-time use.

joris57, 2018-04-26:
Use case: I recently equipped my delta with an eccentrically-placed inductive probe and have been testing with G30 the results of G33 calibration.

Although the results looked ok, when I started a print, I noticed strange behavior of the effector.

I guess you can understand that it took me HOURS of testing, software reconfiguration, regression testing, ... to finally find out that after having performed a G30 previously and when moving the effector in certain directions with the z-probe triggered, this had an effect on the movement (or at least the internal step counters). Far from obvious ... 

What I certainly find strange (and inconvenient) is that this G30 issue is persistent through homing (it does not persist after reboot). 

Although there seems to be a simple workaround : reboot and do NOT execute G30 before printing (which I will test this morning), I hope you will agree with me that executing a G30 should not have this effect on the stability of Marlin?

thinkyhead, 2018-04-27:
It certainly is a weird side-effect. It reminds me of the strangeness we sometimes see if `G28` is forgotten before trying to move, or trying certain moves when an endstop is triggered.

The behavior of `G30` is certainly open to reconfiguration. It's not the most popular command, so changing it wouldn't be too disruptive. The simplest change we could make is to have `E1` be the default, so it always engages and disengages the probe unless `E0` is specified. That seems like it would be a saner default.

joris57, 2018-05-02:
I looks like G30 E1 does not show this strange behavior.

For my eduation: can you explain what is meant by 'engaging' in this content. I understand that the probe is deployyed, but there seems to be more to it as it seems to impact some parameters if he probe gets triggered during a move in that state?

thinkyhead, 2018-05-03:
> it seems to impact some parameters if the probe gets triggered during a move in that state

The only extra effect I can guess is that the probe remains enabled.
You might also see the same strange behavior just using `M401`.

joris57, 2019-01-24:
Don't have the hardware any more to further follow up on this.

closed 2019-01-24 by joris57