MarlinFirmware/Marlin #10526: Bilinear bed leveling works unproperly after update

c0m1c5an5, 2018-04-25:
### Description

After updating Marlin bilinear bed leveling started behaving wiredly. While the mesh seems to be acceptable for me, the actual printing results are kind of bad. It took me quite a long time to test all that could go wrong, but I did not manage to find a solution so far. Before the update all worked nicely, but at the moment I am not even sure thet this is a program error. ( 3d printer model - Anet A8 )

### Problems tested

1. Bad probe - replacing did nothing.
2. High probe deviation - results are quite nice.
3.  Z-axis not aligned - tested

### My configuration file

https://pastebin.com/UpzB7UZb

### Leveling mesh

-0.669  -0.049  +0.039  -0.451
-0.818  -0.139  +0.021  -0.324
-0.940  -0.248  -0.030  -0.281
-1.147  -0.300  -0.006  -0.128

### The actual performance

[image]

As you can see there are some squares that were printed too high, some printed too low and only one was printed rather nicely.

Roxy-3D, 2018-04-25:
I'm biased on this point...  But I think more mesh points would be helpful.   Perhaps 8 x 8 or 10 x 10 ?

But before you do that...   Do a G26 and post a picture of the pattern.   That will tell us how accurate your current mesh points are with regard to the bed height.

c0m1c5an5, 2018-04-25:
Yet G26 gives quite a nice print
[image]

Roxy-3D, 2018-04-25:
OK...  

That does look good.   Are you using a very recent version of Marlin?   Those partial circles on the G26 print were fixed a few weeks ago?    It shouldn't matter for what we care about.   But it is helpful to be using the latest bugfix version when trying to work through problems.

From looking at the G26 pattern, it does look like the front left is not the same as the rest of the bed (with regard to how the first layer is going down).   It is hard to tell but I think it would be helpful to raise the 3 mesh points in the front left corner by .05mm or .1mm.    Can you try raising those mesh points by .05mm, save the mesh, and restart the print.   If it is better, but still not right, try raising those mesh points by another .05mm and see if that is better.

One more suggestion.   Do you have  #define DOUBLECLICK_FOR_Z_BABYSTEPPING turned on?   That is very helpful to fine tune the nozzle height while your print is drawing the skirt on the first layer.   You might want to try that.

c0m1c5an5, 2018-04-25:
I am not currently using the newest possible version, however when I was trying to fix the problem myself I did try using the latest RCbugfix (that did not help). But it seems like the problem is starting to get clearer. How could G26 print all nicely while the usial print made such a horrible results? It seems like the actual printing from the sd creates that misalignment, but how is that possible? (My startup script contains only G28 and G29). And I would also like to find out how do I edit the mesh if ubl is not turned on?

Roxy-3D, 2018-04-25:
> How could G26 print all nicely while the usial print made such a horrible results?

Unknown...   

> And I would also like to find out how do I edit the mesh if ubl is not turned on?

 M421 - Set a single Z coordinate in the Mesh Leveling grid. X<units> Y<units> Z<units> (Requires MESH_BED_LEVELING or AUTO_BED_LEVELING_UBL)

c0m1c5an5, 2018-04-26:
> M421 - Set a single Z coordinate in the Mesh Leveling grid. X Y Z (Requires MESH_BED_LEVELING or AUTO_BED_LEVELING_UBL) 
I have neither of them on because I can not use bilinear and mesh leveling simountaniously and UBL takes too much space in ROM (40k over the limit).

Here is a 7x7 leveling grid:

-0.847 -0.410 -0.103 +0.095 +0.073 -0.082 -0.266
-0.909 -0.450 -0.144 +0.071 +0.081 -0.058 -0.230
-0.994 -0.520 -0.205 +0.028 +0.063 -0.081 -0.206
-1.067 -0.593 -0.262 -0.008 +0.049 -0.028 -0.159
-1.138 -0.666 -0.325 -0.047 +0.026 -0.044 -0.095
-1.194 -0.722 -0.351 -0.057 +0.031 +0.000 -0.015
-1.355 -0.728 -0.370 -0.053 +0.073 +0.068 +0.097

thinkyhead, 2018-04-26:
@c0m1c5an5 — Watch out for imperfections in the X gantry. Even a small twist can mess up the mesh in ways that the firmware can't work around. Fortunately with UBL you can do manual probing of points using the nozzle and then compare to the inductive probe results.

thinkyhead, 2018-04-27:
The key word in my explanation is _**twist**_. You can _**slant**_ your X-gantry and no problems will arise. But a twist in the gantry causes the angle of the probe to change as the carriage moves in the X, and this slightly affects the point on the bed that it's probing — either more forward or more back, and this introduces a small error in the distance measured.

A manual probe of the bed with the nozzle, using the paper test, will produce a grid that you can compare to the grid you get from probing with the sensor. You will likely find that the manual probed grid is very effective, and you will be able to tell by comparing them just where the errors are arising due to the sensor being off-angle.

We've seen this many times before, and various adjustments are known to help. Make sure your X belt isn't too tight. Make sure that the wires and bowden tube aren't pulling the X carriage out of alignment. And so on. Basically, make all necessary adjustments so that the sensor is vertical to the bed at all times and that nothing can throw it off. And of course, do all probing with the bed and nozzle heated to printing temperature.

c0m1c5an5, 2018-04-27:
> We've seen this many times before, and various adjustments are known to help. Make sure your X belt isn't too tight. Make sure that the wires and bowden tube aren't pulling the X carriage out of [...]

Quite a nice suggestion but not the cause of my problem sadly.

Roxy-3D, 2018-04-27:
Can we try bumping the mesh dimensions up to 10 x 10 and rebuild the mesh?    With a higher resolution mesh, the G26 pattern will allow us to fine tune things.

c0m1c5an5, 2018-04-27:
Same as 4x4 grid above G26 with 10x10 prints rather nicely, however lines are thicker from left to right and from top to bottom. But that is not a huge problem, the one is that the actual 3d models are printed incorrectley.

[image]

Roxy-3D, 2018-04-27:
I think I would raise these mesh points by .05mm and see how things look.  It looks like the nozzle is closer to the bed here than in other locations:

[image]

c0m1c5an5, 2018-04-28:
> I think I would raise these mesh points by .05mm and see how things look. It looks like the nozzle is closer to the bed here than in other locations

> lines are thicker from left to right and from top to bottom

Yep, that is what I was talking about.

But how do I edit mesh if:

> M421 - Set a single Z coordinate in the Mesh Leveling grid. X Y Z (Requires MESH_BED_LEVELING or AUTO_BED_LEVELING_UBL)

I have neither of them on because I can not use bilinear and mesh leveling simultaneously and UBL takes too much space in ROM (40k over the limit). The config  I have now takes 98% of available flash space.

Roxy-3D, 2018-04-28:
The comment is wrong...  You can adjust mesh point values with M421 if you Bi-Linear enabled:
```

#elif ENABLED(AUTO_BED_LEVELING_BILINEAR)

  /**
   * M421: Set a single Mesh Bed Leveling Z coordinate
   *
   * Usage:
   *   M421 I<xindex> J<yindex> Z<linear>
   *   M421 I<xindex> J<yindex> Q<offset>
   */
  inline void gcode_M421() {
    int8_t ix = parser.intval('I', -1), iy = parser.intval('J', -1);
    const bool hasI = ix >= 0,
               hasJ = iy >= 0,
               hasZ = parser.seen('Z'),
               hasQ = !hasZ && parser.seen('Q');

    if (!hasI || !hasJ || !(hasZ || hasQ)) {
      SERIAL_ERROR_START();
      SERIAL_ERRORLNPGM(MSG_ERR_M421_PARAMETERS);
    }
    else if (!WITHIN(ix, 0, GRID_MAX_POINTS_X - 1) || !WITHIN(iy, 0, GRID_MAX_POINTS_Y - 1)) {
      SERIAL_ERROR_START();
      SERIAL_ERRORLNPGM(MSG_ERR_MESH_XY);
    }
    else {
      z_values[ix][iy] = parser.value_linear_units() + (hasQ ? z_values[ix][iy] : 0);
      #if ENABLED(ABL_BILINEAR_SUBDIVISION)
        bed_level_virt_interpolate();
      #endif
    }
  }
```

thinkyhead, 2018-04-28:
Comment is now fixed.

c0m1c5an5, 2018-04-29:
I have just discovered an interesting fact. If you compare these two meshes:

-0.815 -0.335 +0.054 +0.020 -0.239
-0.886 -0.366 +0.014 +0.027 -0.195
-0.968 -0.446 -0.005 +0.048 -0.100
-1.022 -0.479 -0.019 +0.086 +0.017
-1.045 -0.475 +0.008 +0.198 +0.212

and

-0.260 -0.049 +0.047 -0.247 -0.778
-0.319 -0.081 +0.030 -0.249 -0.739
-0.389 -0.135 +0.013 -0.208 -0.631
-0.447 -0.172 +0.008 -0.150 -0.498
-0.455 -0.172 +0.035 -0.058 -0.319

You can see that I have lifted both left corners up (hoping that making all as flat as possible will help).
However when I compared two test prints from both meshes I noticed that they are practically identical.

[image]

I have marked the squares with colors:
green - extruder too high
yellow - nicely printed
red - extruder too low

Just some more information on the problem.

thinkyhead, 2018-04-30:
If the problem is in the angle of the X carriage / probe as it traverses the gantry then adjusting the bed will not change the results. Again, the evidence suggests that something in your physical setup is skewing the probe results. According to the various reports we've received on this kind of behavior, sometimes it can help to adjust the belt tension.

c0m1c5an5, 2018-04-30:
> If the problem is in the angle of the X carriage / probe as it traverses the gantry then adjusting the bed will not change the results. Again, the evidence suggests that something in your physical [...]

I have just checked both x and y axis. Sadly they both have almoust no deviation between the ends, nothing is twisted or misaligned, belt tension is acceptable and reducing it does nothing. So the problem must be somewhere else. However I do completely agree that it is something mechanical. May be some other suggestions?

c0m1c5an5, 2018-05-02:
New update!

Recently I have found out that my printer has been standing on an awfully unleveled surface. So i made a leveling platform for it. Now all is perfectly aligned as you can see on the picture following
[image]
Laser level was used to do show how perfectly all the axis align. However the most sad part it the fact that those changes did ABSOLUTELY NO thing to help with unproper bed leveling =(.

thinkyhead, 2018-05-25:
Sorry we can't be more help. Unfortunately there's nothing actionable here, and we're too busy with development to do support of this kind. For faster responses and a larger pool of helpful and experienced users, you should take advantage of the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)

closed 2018-11-07 by thinkyhead