MarlinFirmware/Marlin #10527: [bugfix-2.0.x] Strange Probing pattern UBL

boelle, 2018-04-25:
After much time i got my config ported from Repetier and got pid tune etc done

Found the steps for UBL here: http://marlinfw.org/docs/features/unified_bed_leveling.html

but the bed is probed in a VERY strange pattern, i have attached my current config and would be very happy if somone could have a look at say if the config is correct or at least tell me what i can expect so i can tell if what is see is "normal"

Q: how much does the Z_PROBE_OFFSET_FROM_EXTRUDER matter when i home to a max/top mounted Z endstop? that is the only one i have not spend much time on verifying as correct

Roxy-3D, 2018-04-25:
UBL tries to probe outwards from the starting point.   The reason for this is because the further the nozzle gets from the starting location, the more likely there isn't enough clearance.   Especially on poorly calibrated Delta's this is helpful because of the cup and bowl problem.   If the nozzle is getting too close to the bed, this gives the user time to stop the probing before damage happens.

It is important to have Z_PROBE_OFFSET_FROM_EXTRUDER set accurately.   The bed leveling will work even with a poorly set Z_PROBE_OFFSET_FROM_EXTRUDER.   But the problem is the first 10mm (Fade Height) of layers are squished or expanded such that all leveling is turned off after this height.  Having a poorly calibrated Z_PROBE_OFFSET_FROM_EXTRUDER will affect the printed parts Z Dimension.

boelle, 2018-04-25:
btw.. i have set fade height much lower, 1mm as the bed is not that bad

Roxy-3D, 2018-04-25:
> what are the most precise way of getting the Z_PROBE_OFFSET_FROM_EXTRUDER?

Home the machine...   And verify that when the nozzle just touches the bed (at the home location) the Z-Height is very close to 0.0 mm.     (Via LCD or M114)

I usually do that.   But I also tweak it with the Z-BabyStepping while the skirt is being drawn for the part.

> btw.. i have set fade height much lower, 1mm as the bed is not that bad

I would suggest leaving it at 10 mm until you see how the prints turn out.   Most people never change it from that value.

boelle, 2018-04-25:
the tricky part is that my machine homes up (max endstop)

so i have 2 numbers to get right Z_MAX_POS and Z_PROBE_OFFSET_FROM_EXTRUDER

will leave it at 10mm :-) thanks for the help so far

boelle, 2018-04-25:
But i guess i dial in Z_MAX_POS so that Z is 0 when a heated nozzle is a sheet of paper from touching the bed (0.1mm)

then go up until probe does not trigger anymore, i use an infrared probe so its triggered at this point but not touching the bed. the distance i have to go up must be Z_PROBE_OFFSET_FROM_EXTRUDER... correct?

Roxy-3D, 2018-04-25:
Yeah...   Homing to Z-Max does add an extra variable.   And the Z_PROBE_OFFSET_FROM_EXTRUDER number doesn't have to be perfect.   But the closer you get it to exactly right, the less other issues you will have.   Z-BabyStepping helps a lot to work around errors in that number (or big changes in room temperature, etc.)

Z_PROBE_OFFSET_FROM_EXTRUDER is defined as the height at which the probe triggers.    Where ever it triggers...  If you move the nozzle to 0.0 mm,  the nozzle should just barely touch the bed.

boelle, 2018-04-25:
An idea (not sure if its worth 0.01$) but could this be automated some way so that user issues a command and then lower the nozzle until it touches the bed and when its at the bed enter another command and it figures out both numbers? just a thought

boelle, 2018-04-25:
last and then i think we can close this one again

if time where not a problem and i wanted to push the ramps/mega combo to the extreme when it comes to UBL are there tweaks i can do to the code? ie more probes etc

Roxy-3D, 2018-04-25:
Not really...  The biggest thing you can do is set the mesh size to 10 x 10 or 12 x 12.   And then use G26 as a basis to edit the mesh so it perfectly reflects reality.   If the mesh is accurate...   You should get 100% adhesion across the entire bed every time.

boelle, 2018-04-25:
oki, thanks for all the help

boelle, 2018-04-26:
hope you dont mind i ask if i do it correct, but are these steps correct?

M502
M500
M501
G28
G29 P1
G29 S1
G29 F 10.0
G29 A
M500

The steps i found here http://marlinfw.org/docs/features/unified_bed_leveling.html
also mentions the use of 3 point probe to transform the stored mesh. But i'm in doubt what it will do and if what that does will improve things

boelle, 2018-04-26:
pic of the first test: https://photos.app.goo.gl/NH6nBRO8dwhftpgh1

i have now reduced the 10mm from edge to 5 and doing another probe and test print

boelle, 2018-04-26:
pic of the G26 pattern

https://photos.app.goo.gl/3SL0vXbzn2LlXEV13

all except the last 3 rows at the front are a bit to high up, i would guess the 0.55mm i have set as probe offset

the last 3 rows are way to close an are causing skips in the extruder, BUT this is also where the bed was not probed at all. the rest of the bed is now probed right to the edge.

i did sit at the printer for the whole probing and it does go straight to the edge for left/right and back, but not front

Roxy-3D, 2018-04-26:
I would move the nozzle to the front row and do a G29 P4 R.     That will let you raise those mesh points.   I'm guessing, but raising them .15mm would be a good start.    The nozzle will dance around (expanding outward from the starting location).    If the nozzle positions itself over a mesh point that looks OK, just click and don't edit.        After you get 3 or 4 points along the front row, you can terminate the G29 P4 R (by pressing and holding the encoder wheel) and move the nozzle further over on the row to repeat the operation.

gloomyandy, 2018-04-26:
Isn't the Z offset normally negative (in the configuration file)? A positive value would indicate that the probe trigger point would have the nozzle below the bed?

boelle, 2018-04-26:
the thing i dont understand is that those 3 rows are also there it did not probe the bed

and i have just messured and sure enough there is 23mm between the center of the good mesh point to the last one at the Y0 line on the bed

some something is telling it not to probe the last 23mm in the y direction, i just dont understand why

Roxy-3D, 2018-04-26:
The G29 P4 will probe all locations that the probe can reach.   But some of the bed's mesh points are not reachable.    Those mesh points are invalid and denoted with a '.' in the mesh map.    Those locations can be filled in with manual probing (G29 P2) or with calculated values via G29 P3.     Either way...   Once they are filled in, they are active and a G26 can show how accurate those points are.    Using that information, G29 P4 R lets you edit the flawed points so they are more accurate.

boelle, 2018-04-26:
> Isn't the Z offset normally negative (in the configuration file)? A positive value would indicate that the probe trigger point would have the nozzle below the bed?

ahhh, yes. i guess its how you understand it:

> #define Z_PROBE_OFFSET_FROM_EXTRUDER 0.55   // Z offset: -below +above  [the nozzle]

i put it in as positive as the probe sits just above the nozzle, but yes the trigger point is below the nozzle and that is what should count. correcting that now

boelle, 2018-04-26:
> But some of the bed's mesh points are not reachable.

the probe can reach off of the bed, it can even go over all 4 edges

and its strange that it stops probing excatly 23mm from the front of the bed. The same 23mm is also the Y offset of the probe... so maybe i have entered that wrong too?

Roxy-3D, 2018-04-26:
Very literally...   The Z_PROBE_OFFSET_FROM_EXTRUDER number is the distance between the trigger point and the nozzle.   If the nozzle is at +10.0 mm when the probe triggers...   The Z_PROBE_OFFSET_FROM_EXTRUDER number is -10.0mm.   That is because it is -10.0mm below the nozzle.

boelle, 2018-04-26:
hmm... one thought

it can mechanical reach everywhere

but maybe the config prevents it to move the nozzle past the Y0 line.... that is needed for the probe to reach the Y0 Line

will check that next

Roxy-3D, 2018-04-26:
> it can mechanical reach everywhere

That allows you some extra flexibility (and complexity).   You can define the bed size to be the entire area the nozzle can move around in.    And then you define a MESH_INSET to clamp that area down to just where the bed is.     And if you need a non-semetrical inset, you can adjust the 4 different sides of the print bed in Configuration_adv.h

I do that on my large gMax printer.   I home X & Y off the physical bed.   And I clamp the usable area down to where the bed physically exists using the 4 INSET controls in Configuration_adv.h.

With the printer configured that way, you don't end up with any unprobed mesh points at the end of the auto probing.

boelle, 2018-04-26:
Thanks for all the help

the G26 pattern now looks ok and even for all the mesh points

once the final G26 is done i will do another of those full plate test and make the last adjustments to the probe offset

closed 2018-04-26 by boelle