MarlinFirmware/Marlin #10529: [Bugfix-1.1.x] G30 ignores Z_AFTER_PROBING

joris57, 2018-04-25:
On a delta, Z_AFTER_PROBING is set to 5 `#define Z_AFTER_PROBING             5`

However, the nozzle remains at the level the Z-probe is triggered and does not lift again.

thinkyhead, 2018-04-26:
What about `G30 E1`?

joris57, 2018-04-26:
G30 E1 seems to take into account Z_AFTER_PROBING. Thanks for pointing this out.

This then seems to me like a regression (and documentation) issue. G30 default behavior previously took into account Z_CLEARANCE_BETWEEN_PROBES. With the introduction of Z_AFTER_PROBING, I would have expected that, with the same logic as before, Z_AFTER_PROBING would be applied by default. As far as I'm concerned, his issue can be closed.

closed 2018-04-27 by thinkyhead