MarlinFirmware/Marlin #10531: TMC262 BOB40 support and code changes

chriscg9, 2018-04-26:
TMC262BOB40 does not  to function.
Link to product page
https://www.digikey.com/product-detail/en/trinamic-motion-control-gmbh/TMC262-BOB40/1460-1243-ND/8037670
When configured in the firmware the drivers do not work

I decided to do some digging and found somethings that once changed allowed this driver to function.
stepper_indirection.cpp
Was (Enable is done via spi not pins so why is enable pin set here?)
#define _TMC_DEFINE(ST) TMC26XStepper stepper##ST(200, ST##_ENABLE_PIN, ST##_STEP_PIN, ST##_DIR_PIN, ST##_MAX_CURRENT, ST##_SENSE_RESISTOR)
Fixed (changed the order and switched enable with CS)
#define _TMC_DEFINE(ST) TMC26XStepper stepper##ST(200, ST##_CS_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_MAX_CURRENT)

With this changed i noticed that once the board booted up the motor would enable and hold rotor position and when told to move it would stop holding rotor position. So it would seem the logic is reversed for enable in SPI or just this BOB is different than the normal 26x boards. 

I found what i was looking for and changed them see comment for what it was
#elif HAS_X_ENABLE //1134 find my changes
  #define  enable_X() X_ENABLE_WRITE( !X_ENABLE_ON) //X_ENABLE_ON
  #define disable_X() do{ X_ENABLE_WRITE(X_ENABLE_ON); axis_known_position[X_AXIS] = false; }while(0)//!X_ENABLE_ON

So i am unsure if the changes i have made will have any ill consequences down the road.
But for the moment i am able to run the motors and power some very big steppers considering i get a full 2.8A at 40V

Any thoughts on the changes? 

Picture of rough setup for bench testing before being put into a Stratasys FDM 
[image]

thinkyhead, 2018-04-28:
I think you're onto something. This is the constructor for `TMC26XStepper`…
```cpp
TMC26XStepper(int number_of_steps, int cs_pin, int dir_pin, int step_pin, unsigned int current, unsigned int resistor=150);
```

thinkyhead, 2018-04-28:
> Any thoughts on the changes?

The initializer change in `stepper_indirection.h` fixes a bug, so that's a good start!

But the change to `Marlin.h` is not needed, because we have these in `Configuration.h`:
```cpp
// For Inverting Stepper Enable Pins (Active Low) use 0, Non Inverting (Active High) use 1
// :{ 0:'Low', 1:'High' }
#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define E_ENABLE_ON 0 // For all extruders
```
…and they can just be set to 1 to invert the logic.

If the entire TMC26x line consistently inverts these pins, then this is something we should alert users about in the sanity-check file. Otherwise, we'll just have to document this consideration in a comment above the `HAVE_TMC26X` option in `Configuration_adv.h`.

I'll patch the first issue in the bugfix branches now, and then check the datasheets to see what we need to do about the second issue.

chriscg9, 2018-04-29:
Awesome.
Would it be better i just fix my code or download the bugfix branch?

Thanks for your help.
I did not see the invert enable when skimming the config file thanks.

Do you have any recommendations as to what to set the microstep amount?
Will i be limited in my stop speed by setting it to the full 256? vs setting it to say 64?
Or will the top end not be noticeable? 
My machine does have a very heavy XY gantry so i cant go to any crazy speeds.

Not my picture but this is what my gantry looks like
[image]

chriscg9, 2018-04-30:
Okay thanks for the advice on the steps.

What board would be able to handle the full 256 steps and still ramps compatible? 
Re-arm or due? 
Any one better than the other?

thinkyhead, 2018-04-30:
Due runs 4-5 times faster than AVR. Re-Arm runs 6-7 times faster. Either one should be more than capable of keeping up with 256 real micro-steps.

Note that Due is not RAMPS compatible. It requires a RAMPS4DUE with 3.3v support.

boelle, 2019-02-20:
@chriscg9 is this one "solved" ?

chriscg9, 2019-02-20:
Yes it is solved.

closed 2019-02-20 by chriscg9