MarlinFirmware/Marlin #10533: [2.0.x] RAMPS FD V2.1 motor vibrate when machine starts up if EEPROM enabled

kevintong116, 2018-04-26:
I am now using RAMPS FD V2.1  with due
If I enable the EEPROM function, motor vibrate for a moment when the machine starts up/ being reset
This does not happen if EEPROM not enabled

What is the reason?

configuration.h is attached
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1951566/Marlin.zip)

thinkyhead, 2018-04-28:
Hard to say. The EEPROM does some extra initialization based on stored values. Try `M502`, `M500` and see if it behaves better on the next reboot.

kevintong116, 2018-04-29:
M502, M500 does not help

thinkyhead, 2018-05-23:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands.

boelle, 2019-02-20:
@kevintong116 did you test with the latest `bugfix-2.0.x` ?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead