MarlinFirmware/Marlin #10538: [bugfix-2.0.x]　G29 BILINEAR can not be executed. Get an out of bounds error.

JXPA, 2018-04-26:
Hello.
I attempted to run G29 for the first time in marlin 2.0, but an error occurred and it was not executed.
This is the log when G29 is executed.
```
: N654 M111 S247*103
: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
: N655 G29*20
: echo:N655 G29*20
: current_position=(0.00, 0.00, 10.00) : >>> G29
: Machine Type: Delta
: Probe: FIX_MOUNTED_PROBE
: Probe Offset X:0 Y:0 Z:0.09 (Aligned With & Above Nozzle)
: Auto Bed Leveling: BILINEAR (disabled)
: ? (L,R,F,B) out of bounds.
```
My understanding, the limit position of the probe is automatically calculated based on　DELTA_PRINTABLE_RADIUS and MIN_PROBE_EDGE in Conditionals_post.h.
I have a printing bed with a diameter of 220 mm and set the DELTA_PRINTABLE_RADIUS to 105.0mm.
MIN_PROBE_EDGE is 10mm.
I am using piezo20 for Z-probe, and offset is set to 0.

The probe reach range should be within the printing bed, but an out of bounds error is output and G29 is not executed.
An error is returned even if setting of L, R, F, B_PROBE_BED_POSITION is enabled in the Configuration.h.
In 1.1.X, this kind of trouble never occurred, and I do not know why such an error occurred.

Description of my printer
type : Delta
controller : ArduinoDue(genuine) + Ramps-FD V2.2(DIY) + RepRapDiscount Full Graphic Smart Controller
Motor driver : TMC2130(X,Y,Z,E0)
Z-probe : Piezo20
Connect ArduinoDue programming port and Windows 10.
For compiling and flushing use Arduino 1.9.0-beta.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1953220/Marlin.zip)

thinkyhead, 2018-04-26:
This is a bug in the Marlin bugfix branches. The check for delta bounds is incorrect. I'll post a fix shortly.

JXPA, 2018-04-26:
Thank you for your quick response.

thinkyhead, 2018-04-27:
I've posted a patch to the bugfix branches, so (hopefully?) there will be no more complaints from `G29` about the pre-calculated probe bounds. If it's fixed, go ahead and close this, otherwise hit us up again and we'll kick the code harder.

bobbydamaker, 2018-05-21:
I know that this is closed for 2.0, but I am also experiencing it on 1.1.X Bugfix.  Was the fix published to both?

thinkyhead, 2018-05-21:
Yes it was. Currently on DELTA for ABL it checks that you can reach:
- (left, 0)
- (right, 0)
- (0, front)
- (0, back)

Try passing some values to `G29 Lnn Rnn Fnn Bnn` to see how far from the expected range it's failing for you.

thinkyhead, 2018-05-21:
That makes sense. You gave a probing range of 0 wide and 0 deep.

How wide is your bed?

bobbydamaker, 2018-05-21:
250 Round (125 Radius)  Printable is set at 125.  Calibrate is set at 110.

thinkyhead, 2018-05-21:
So… `G29 L-100.0 R-100.0 F100.0 B100.0` should work (assuming you're not limited by your `MESH_INSET` and `MIN_PROBE_EDGE` settings).

thinkyhead, 2018-05-21:
So… `G29 L-110.0 R110.0 F-110.0 B110.0` should work (assuming you're not limited by your `MESH_INSET` and `MIN_PROBE_EDGE` settings). And, `G29` by itself should also hopefully just do the right thing.

bobbydamaker, 2018-05-22:
The good news is that G29 L-90.0, R90.0, F-90.0, B90.0 finally worked with both the 1.1.X Bugfix and 2.0.x Bugfix.  If I run a G29 alone I get the L,R,F,B is Out of Bounds Error still.

I have attached my configuration file in case that is helpful for troubleshooting.  I am using a modified BIQU Kossel Pro.  I was running successfully with a vanilla install of 1.1.8.  Was trying to get to 1.1.9 or 2.0.x to enable Linear Advance more than anything else.

[Configuration.pdf](https://github.com/MarlinFirmware/Marlin/files/2024636/Configuration.pdf)

thinkyhead, 2018-05-22:
I'll take a look at the config soon… about to retire for the night.

But PDF? That's a first for us here!

You should put `.txt` on the end of the filename or ZIP the configs together to get them to upload, and then we can just use them as they are for testing.

bobbydamaker, 2018-05-23:
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2029293/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2029292/Configuration.txt)

Sorry about that.  I was tired and not thinking.  Here you go.

bobbydamaker, 2018-05-27:
I figured it out.  The offfset of my probe exceeded the MIN_PROBE_EDGE setting on the X plane.  Once I turned those around, we are good to go.  You can keep this closed.

SimonSolar2C, 2018-06-01:
I received the same error, ? (L,R,F,B) out of bounds, using V1.1.x pulled down on 15th May 2018
I only received this error when I moved my probe from left of the extruder (X-53) to right of the extruder(X25). Y is now zero, and MIN_PROBE_EDGE is default 10.
I havent dug any deeper, but I expect that these values are valid. If not, I thought the pre-compiler would catch it before its a problem when running?

SimonSolar2C, 2018-06-01:
Okay I resolved above values by increasing MIN_PROBE_EDGE to 26, ie more than my X_probe_offset.
But I think there is a bug there somewhere if the probe is to the right of the extruder.

thinkyhead, 2018-06-01:
Yes, it would be good to catch this in the pre-compiler rather than have Marlin complaining at runtime. I'll see if it's something obvious and try to patch it now. Meanwhile, @SimonSolar2C can you drop a copy of your `Configuration.h` (with the settings that exhibited the problem) into a reply so we can use it to test our solution? We'll want to make sure it throws an `#error` at compile time.

SimonSolar2C, 2018-06-03:
Sure – attached is the file as it was when generating the problem.

 

Kind regards,

Simon Dyer

bobbydamaker, 2018-06-03:
My probe is also to the right of the extruder, and it seemed that I only had to adjust the L parameter when you had me inputting the G29 (L,R,F,B) values manually.

thinkyhead, 2018-06-03:
> I can no longer get consistency between G28 and G29

If the probe is tilted or the gantry is twisted, even by a small amount, it can produce readings that appear too high. To see if discrepancies exist, you can do a `PROBE_MANUALLY` probe with the nozzle and print out the mesh, then compare that mesh to the automated probing result. If the meshes differ, then it probably indicates some kind of probe mounting issue. If the difference is consistent across the whole mesh, then it simply indicates the Z offset (`M851 Z`) needs adjustment.

SimonSolar2C, 2018-06-05:
@bobbydamaker  & @thinkyhead Thanks for the comments and sorry I went a bit off topic there.  The problem I describe above seems to have corrected itself !
I have this inkling that after a firmware reflash, the mesh is not cleared out during a g28, or maybe accumulates like a moving average. After a few cycles, all is well with the consistency of a g28 versus a g28 - both correctly move to G0 Z0 in the centre of the bed, but not when I first tried it after reflash.

mq269, 2019-03-22:
I'm getting this same error on a 1.1.9 version. ? (L,R,F,B) out of bounds. 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2994624/Configuration.zip)

mq269, 2019-03-22:
I commented out the manual table definitions lines 1040 - 1050, and it is working correctly now.

69kevinutah69, 2019-06-22:
this is still an active issue i just got this error on the 2.0.x bugfix

mylife4aiurr, 2019-06-23:
me too!
Recv: ? (L,R,F,B) out of bounds.

#if ENABLED(DELTA)

  // Make delta curves from many straight lines (linear interpolation).
  // This is a trade-off between visible corners (not enough segments)
  // and processor overload (too many expensive sqrt calls).
  #define DELTA_SEGMENTS_PER_SECOND 200

  // After homing move down to a height where XY movement is unconstrained
  //#define DELTA_HOME_TO_SAFE_ZONE

  // Delta calibration menu
  // uncomment to add three points calibration menu option.
  // See http://minow.blogspot.com/index.html#4918805519571907051
 #define DELTA_CALIBRATION_MENU

  // uncomment to add G33 Delta Auto-Calibration (Enable EEPROM_SETTINGS to store results)
 #define DELTA_AUTO_CALIBRATION

  // NOTE NB all values for DELTA_* values MUST be floating point, so always have a decimal point in them

  #if ENABLED(DELTA_AUTO_CALIBRATION)
    // set the default number of probe points : n*n (1 -> 7)
    #define DELTA_CALIBRATION_DEFAULT_POINTS 7
  #endif

  #if EITHER(DELTA_AUTO_CALIBRATION, DELTA_CALIBRATION_MENU)
    // Set the radius for the calibration probe points - max DELTA_PRINTABLE_RADIUS for non-eccentric probes
    #define DELTA_CALIBRATION_RADIUS 70.00 // (mm)
    // Set the steprate for papertest probing
    #define PROBE_MANUALLY_STEP 0.05      // (mm)
  #endif

  // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
  #define DELTA_PRINTABLE_RADIUS 98.0    // WAS 96.0

  // Center-to-center distance of the holes in the diagonal push rods.
  #define DELTA_DIAGONAL_ROD 228.0        // (mm)

  // Distance between bed and nozzle Z home position
  #define DELTA_HEIGHT 300.00             // (mm) Get this value from G33 auto calibrate

  #define DELTA_ENDSTOP_ADJ { 0.00, -0.64, -0.71 } // Get these values from G33 auto calibrate

  // Horizontal distance bridged by diagonal push rods when effector is centered.
  #define DELTA_RADIUS 108.56              // (mm) Get this value from G33 auto calibrate

  // Trim adjustments for individual towers
  // tower angle corrections for X and Y tower / rotate XYZ so Z tower angle = 0
  // measured in degrees anticlockwise looking from above the printer
  #define DELTA_TOWER_ANGLE_TRIM { -0.96, 0.92, 0.04 } // Get these values from G33 auto calibrate

  // Delta radius and diagonal rod adjustments (mm)
  #define DELTA_RADIUS_TRIM_TOWER { 0.0, 0.0, 0.0 }
  #define DELTA_DIAGONAL_ROD_TRIM_TOWER { 0.0, 0.0, 0.0 }

#define BLTOUCH

#define X_PROBE_OFFSET_FROM_EXTRUDER  9   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -35   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER  0   // Z offset: -below +above  [the nozzle]

// Certain types of probes need to stay away from edges
#define MIN_PROBE_EDGE 10

#define Y_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)
#define Y_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Z_MIN_POS 0
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
#define Z_MAX_POS MANUAL_Z_HOME_POS

#define AUTO_BED_LEVELING_BILINEAR

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
  #define RIGHT_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
  #define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
  #define BACK_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)

thinkyhead, 2019-07-11:
@mylife4aiurr — Comment out these lines:
```cpp
  #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
  #define RIGHT_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
  #define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
  #define BACK_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
```

mylife4aiurr, 2019-07-11:
i got everything squared with bilinear but decided i like ubl better anyway. thanks

Floriantoine, 2019-08-27:
i have this error::? (L, R, F, B) out of range.

yet I have the last vertiy of marlin, the path is already in it I do not understand where it comes from

Ohmarinus, 2019-11-13:
I'm having the same issue. G33 was executed correctly, and saved with M500. Then I use G28 to home the axes (on a Delta printer) and the I send G29. I am using a homebrew Kossel, on my homebrew 'MendelMax 1.6' I can just use G29 without any errors but using a different version of Marlin.

M665 settings:
`M665 B70 H320 L210 R100 S200`

G33 output:
`Calibration OK                                rolling back.
.Height:313.01  Ex:+0.00  Ey:-1.03  Ez:-0.05  Radius:101.32
.               Tx:-0.23  Ty:+0.37  Tz:-0.13
Save with M500 and/or copy to Configuration.h
`

My config files:
[config_adv_g29.zip](https://github.com/MarlinFirmware/Marlin/files/3842541/config_adv_g29.zip)

I hope someone knows the answer.

commit aab4d9e2 (2018-04-27): Fix ABL grid bounds test for Delta/SCARA

Fix #10538