MarlinFirmware/Marlin #10539: Confusing #warning directive when compiling for unsupported AVR

per1234, 2018-04-27:
### Description

When you attempt to compile Marlin for an unsupported AVR microcontroller you get the error:
```
Pins for this chip not defined in Arduino.h! If you have a working pins definition, please contribute!
```
which is produced by this line:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_AVR/fastio_AVR.h#L55

But Arduino.h does not contain pin definitions.

I'm submitting an issue report rather than PR because I'm not sure what would be the most correct wording for the error message. Maybe:
>Pins for this chip not defined in Marlin/src/HAL/HAL_AVR! If you have a working pins definition, please contribute!

If I can get input on this I'm happy to submit a pull request for the change.

### Steps to Reproduce

1. Open the Marlin sketch in the Arduino IDE
1. Tools > Board > Arduino/Genuino Uno (no, I don't expect Marlin to run on ATmega328P, it's just a convenient board selection to trigger the error).
1. Sketch > Verify/Compile

**Expected behavior:** Error message to be accurate.

**Actual behavior:** Error message refers to pin definitions in a file that does not contain pin definitions.

thinkyhead, 2018-04-27:
Marlin's fastio headers define the items starting with `DIO#_`, but they only correlate to the pins that are defined for each MCU by the Arduino headers (included indirectly via `Arduino.h`). For example, take a look at `fastio_1280.h`…
```cpp
#define DIO2_PIN    PINE4
#define DIO2_RPORT  PINE
#define DIO2_WPORT  PORTE
#define DIO2_DDR    DDRE
#define DIO2_PWM    &OCR3BL
```
Marlin doesn't define `PINE4`, `PINE`, `PORTE`, `DDRE`, or `OCR3BL`. The Arduino headers do. And that is what we are asking for with that error message.

per1234, 2018-04-27:
>Marlin doesn't define PINE4, PINE, PORTE, DDRE, or OCR3BL. The Arduino headers do. And that is what we are asking for with that error message.

The problem is not that the `PINE4`, etc. macros are not defined, because of course the equivalent macros are defined for every AVR microcontroller in their ioxxxx.h file. The problem is that the `DIO2_PIN`, etc macros are not defined. To add support the user would need to make a new fastio_xxxx.h file (or use a compatible existing file for their part and add that MCU-specific macro to fastio_AVR.h), not do something to Arduino.h or any file included from Arduino.h.

NOTE:
I noticed that in fact there are pin definitions for ATmega328P but the MCU-specific macro was wrong (https://github.com/MarlinFirmware/Marlin/pull/10540), which caused the error to be displayed when compiling for ATmega328P. So Arduino/Genuino Uno is a bad example to use. I've switched to Arduino Leonardo as an example of a board that would generate the warning.

thinkyhead, 2018-04-28:
What defined identifier(s) should we be looking for to distinguish a Leonardo?

per1234, 2018-04-28:
The MCU on the Leonardo is ATmega32U4 so it's `__AVR_ATmega32U4__`.
They're all listed here:
https://www.microchip.com/webdoc/avrlibcreferencemanual/using_tools_1using_avr_gcc_mach_opt.html
Arduino also has a board identifier macro `ARDUINO_AVR_LEONARDO` if you wanted to specifically identify a Leonardo board.

I'm not sure why that's relevant though. I was just using the Leonardo as an example of a **Tools > Board** menu selection that would trigger the misleading warning so that people could easily reproduce the issue.

thinkyhead, 2018-04-28:
> I'm not sure why that's relevant

Just curious which AVRs are listed that we could still add support for. Obviously not the Leonardo, since it only has 32K.

But it would be nice to expand this section to include some more helpful messages, such as:

**Marlin can't fit on a Leonardo. Select a different board.**

…and then only "request help" for the AVRs that actually are supportable. Of course, this error message is not really a serious request for help, but more of a lark.

per1234, 2018-04-28:
I just did a [parametric search on microchip.com](http://www.microchip.com/paramChartSearch/chart.aspx?branchID=30047) for AVRs with 64+ kB program memory and the only ones not already listed in fastio_AVR.h are ATmega64 (which could be added to the `AVR_ATmega1284_FAMILY`), and ATmega128 (which could be added to the `AVR_ATmega2561_FAMILY`). But I don't trust those results since they are missing the obvious ATmega2560, ATmega1280, ATmega1281. I did a parametric search on Digikey and got many more results:
https://www.digikey.com/products/en/integrated-circuits-ics/embedded-microcontrollers/685?FV=26c00ab%2C26c00ac%2C26c00c8%2C26c00d9%2C26c00e1%2C26c00e8%2C26c00f2%2C26c00f4%2C26c0108%2C26c011e%2C26c0125%2C7e8000e%2Cffe002ad&quantity=&ColumnSort=0&page=4&pageSize=25

thinkyhead, 2018-04-29:
Well, for now it will just say:

**No FastIO definition for the selected AVR Board.**

closed 2018-04-29 by thinkyhead