MarlinFirmware/Marlin #10541: TMC2130 build issues with Marlin 1.1.8

Nicoolai, 2018-04-27:
Hi,
I'm not really sure if I'm doing something wrong here.
I've built Marlin 1.1.8 some time ago, with TMC2130 support, without any issues.
Now, I wanted to make some changes but can no longer build it.
Instead, I get the following error:
\Marlin.ino:66:30: fatal error: TMC2130Stepper.h: No such file or directory

I've checked that the library is indeed install, in the arduino library manager.
I've tried to uninstall the arduino IDE, remove all folders from it and reinstall. Same issue appears.

Any ideas?

thinkyhead, 2018-04-28:
Don't touch the "Include Library…" menu before building. It modifies your `Marlin.ino` and is causing the error. Just open the project and build.

Nicoolai, 2018-04-29:
I haven't touched the includes. I have only gone into the library manager, and ensured that TMC2130 was installed.

It fails on the
```
#if ENABLED(HAVE_TMC2130)
  #include <SPI.h>
  #include <TMC2130Stepper.h>
#endif
```
Even though TMC2130 is installed, in the library manager, it seems to not be able to find it :(

Nicoolai, 2018-04-29:
Tried both bugfix branches, same result.
```
Build options changed, rebuilding all
In file included from sketch\stepper.h:48:0,

                 from sketch\Marlin_main.cpp:256:

sketch\stepper_indirection.h:57:30: fatal error: TMC2130Stepper.h: No such file or directory

   #include <TMC2130Stepper.h>

                              ^

```
Is this only happening for me? Is my setup possibly incorrect somehow?
All I did was pull the branch, remove the comments on `#define HAVE_TMC2130` in configuration_adv.h and hit the verify button.

Nicoolai, 2018-04-29:
So, I don't know what is going on.
I tried to manually download the latest version of TMC2130Stepper and drop it in my libraries folder.
It compiles now.
Before I was simply using the Library Manager tool, that downloads and updates.

thinkyhead, 2018-04-30:
It’s hard to say. Possibly the library manager was confused. Glad you got it working!

closed 2018-04-30 by Nicoolai