MarlinFirmware/Marlin #10543: [2.0] Build Fail for Azteeg X3 Pro

TGMods, 2018-04-27:
Hi guys, 
I've been using Marlin-bugfix-2.0.x for a while now on my HyperCube powered by a Re-ARM without any problems. It was also working on my Azteeg X3 Pro powered D-Bot without any problems until today.

The other day I modified my D-Bot and that required a change to my configuration.h file. I thought it also made sense to upgrade to the latest version of Marlin 2.

This is where my troubles began. I've been using PlatformIO as it is the only way I can program my Re-ARM. I also used it to program the X3 Pro. I have no problem building Marlin 2 for the Re-ARM, but, building for the X3 Pro gave me major problems. After a lot of trial and error I discovered the problem, it is in the lib_deps section of latest platformio.ini.

I believe it all stems from the use of library .zip files as when I changed the platformio.ini for an old version that imports the libraries as .git files it builds for my X3 Pro without any problem. 

I know I could just use Arduino IDE, and I did to prove my config files were correct, but I find that it takes a lot longer to build and upload than it does using Platformio.

I get the same issue whether I use Atom or Visual Studio Code. I am using the latest issue of PlatformIO as I deleted and re-installed Atom and PlatformIO.

I have included a zipped copy of the platformio.ini files I used.

Does anyone know of a solution to my problem?

I have also raised this issue with PlatformIO.

[platformio.ini.zip](https://github.com/MarlinFirmware/Marlin/files/1954912/platformio.ini.zip)

thinkyhead, 2018-04-28:
We use the zip archive files to prevent complaints from PlatformIO on Windows, where `git` is not installed, and to relieve users of the burden of having to figure out how to install it.

Make sure to delete the `.piolibdeps` folder before trying to re-build. The first build attempt after deleting that folder may fail, but the next one should succeed.

TGMods, 2018-04-28:
I was given the same information by Ivan Kravets. I've deleted `.piolibdeps` 5 times and every time PlatformIO complains about the same line .
```
#  https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip
  https://github.com/trinamic/TMC26XStepper.git
```
Changing the one entry allows PlatformIO to build Marlin 2.0.x. For some reason I can't figure out PlatformIO on both my PCs doesn't like the particular .zip file.

thinkyhead, 2018-04-29:
The other reason we needed to use [the commit ID](https://github.com/trinamic/TMC26XStepper/commits/master) was that, after switching to the ZIP archive method (using https://github.com/trinamic/TMC26XStepper/archive/master.zip at that time), then the `lib_ignore` lines didn't work (Of course, in that case `lib_ignore master` worked). The fact that `lib_ignore master` worked was a clue. Finally everything began to work when we could give the folders in `.piolibdeps` unique identifiers — the commit IDs — and refer to them in `lib_ignore`.

So, we're stuck with having to choose among one of these problems:
- Users get a confusing message and we have to deal with support requests and repeatedly explain how to install git on Windows.
- Some 32-bit targets break because PlatformIO compares the `lib_ignore` string against the folder names in `.piolibdeps`, and has a bug in doing that.
- …and now suddenly PlatformIO gets confused about ZIP archives.

In all cases, it's really an issue with PlatformIO, not with Marlin. The ZIP archive method is _supposed_ to work as documented. And the `lib_ignore` lines are supposed to work. And we really want to avoid using .git links.

TGMods, 2018-04-29:
Thanks for the explanation @thinkyhead.

I have no trouble with building for my Re-ARM, so I'll keep trying to build for my X3 Pro when necessary and if required, using my workaround until PlatformIO gets sorted out.

closed 2018-05-24 by TGMods