MarlinFirmware/Marlin #10544: Need very slow axis speed for laser, Gcode F command below 10 not recognised ?

mr16k, 2018-04-27:
I have converted a CR-10 and Duplicator + to Laser etch/cutting.
Works fine on most materials, but now want to etch anodised Aluminium.
Laser is 500mW, so need very slow X,Y axis speeds . I am using mm/min
However any F rate below about 10 fails to run XY axis any slower.
Is there some lower limit in Gcode, I could do multiple passes at F 10 but believe I would not get same results as being able to run at say 1 or 2mm/minute.
Any help would be appreciated.

GMagician, 2018-04-27:
I checked code and it seems there is no limit below 10mm/min in parser, F value is converted to numeric value using strtod. 
But there is 'DEFAULT_MINIMUMFEEDRATE' (default is 0) in configuration_adv.h and value is also store in eeprom and can be changed using M205 using S parameter.
You can try with M205 followed by M500 if you think that stored value is not 0

mr16k, 2018-04-27:
Hi,
GMagician,

Thanks for the tip.
I tried in code
M205 S1
M500.
CR-10 made no difference.
Duplicator Plus, Replicator Gcode to X3g conversion does not recognise M205 or M500 code.

Should say I am running both printers off PC. i.e. Loading both printers via SD card. So hacve no direct control from PC and  cannot view velocity vmin.

GMagician, 2018-04-27:
I checked new 2.0 code... what version are you using?

GMagician, 2018-04-27:
Also M503 should report min value. Maybe you have an hardcoded default different from current firmware ones

p3p, 2018-04-27:
There are hardware limits, AVR uses 16bit timers @ 2Mhz for steps, overflow in 33ms, minimum step rate of 30 steps/second, I only had a quick look at the maths but this appears to be enforced  in `calc_timer_interval(rate)` function where if rate < F_CPU / 500000 (32) then it is set to 0. This doesn't quite line up with your F10 though, @ 80 Steps per mm (CR10 default config) the minimum feed rate should be F24.

mr16k, 2018-04-27:
Thanks for your input. 

GMagician
Not sure which version CR-10 is running will try to see next week. I confess I have never connnect the CR-10 via USB, only ever transferred gcode via SD card!

p3p.
You may have hit the nail on the head, I said below F10 but must admit the next lowest speed I tried was F50.
I will load some increments in Gcode just above and below F24 to see where the minimum speed  sets in.

Have a good weekend

cheers

mr16k, 2018-04-28:
p3p,
I guess you are right
Just tried  50, 30, 24, 20, 15 mm/min etching 24mm line. Times agree down to 24mm/min, below that both 20 & 15mm/min show same time as 24mm/min.

Thanks for the explanation.

cheers

thinkyhead, 2018-04-29:
The simplest solution is to get stepper drivers that can do finer micro-stepping and simply increase the number of micro-steps for the steppers that need to move slowly. This would sacrifice higher speeds on AVR for 3D printing, but it would be ideal for a laser engraver.

closed 2018-04-29 by thinkyhead