MarlinFirmware/Marlin #10545: [bugfix-2.0.x] load script makes printer reset

boelle, 2018-04-27:
This short Gcode makes the printer reset

G90 ; absolute positioning
G92 E0 ; zero the extruded length
M109 S225 ; wait for extruder temp
G1 E100 F1800 ; load 100mm Fast
G1 E200 F1800 ; load 100mm Fast
G1 E300 F1800 ; load 100mm Fast
G1 E400 F1800 ; load 100mm Fast
G1 E500 F1800 ; load 100mm Fast
G1 E600 F1800 ; load 100mm Fast
G1 E700 F1800 ; load 100mm Fast
G1 E800 F1800 ; load 100mm Fast
G1 E900 F100 ; load 100mm Slow
M104 S0 ; extruder off
M84 ; disable motors

i use it to load filament from octoprint

Bob-the-Kuhn, 2018-04-27:
No error messages?

Please zip & post the log.

boelle, 2018-04-27:
Nope no log

Bob-the-Kuhn, 2018-04-27:
Yes, no log if printing from the SD card.

Do you have both the LCD & host connected?  

I thought we had prints being done for every source of resets.

boelle, 2018-04-27:
only LCD and octoprint

Prints? this is just a script to heat up and unload the filament... no printing is being done

that is what is strange to me, i use ocotprint to heat up the nozzle to 225 and then i send G1 E100 F1800 from the terminal... the printer resets

and there is no error in the terminal

boelle, 2018-04-27:
so just sending these 2 lines from the terminal in octoprint is enough to make it reset

M109 S225
G1 E100 F1800

last one is send when temp is at 225

thinkyhead, 2018-04-28:
Your E micro-steps of 32x is the problem, as we discussed.
Reduce them or go slower on E and the board will not reset.

closed 2018-04-28 by thinkyhead

boelle, 2018-04-28:
at least i did set the ramps to 16x microstep and of course halfed the steps pr mm

same thing happened so now i will change the script to use half feedrate ie F1800 will be F900

thinkyhead, 2018-04-29:
> would the higher microstep make for better prints in general?

No. An extruder should be totally fine even with 8x or 4x micro-stepping.