MarlinFirmware/Marlin #10546: [FR] G26 should respond with an error when G26_MESH_VALIDATION isn't enabled

Zer0t3ch, 2018-04-27:
Even something as simple as "Not allowed" as a generic error message when receiving any command that wasn't enabled at compile-time would be wonderful. Just spend 15 minutes tearing my hair out because G26 wasn't doing anything and wasn't providing any output other than "OK". (Which I believe happens after batch of serial communications, but still)

GMagician, 2018-04-27:
I'll take a look at it but I think that new features will be only included in 2.0 version

GMagician, 2018-04-28:
@Zer0t3ch it seems that there are some host software compatibility issues adding this feature, read #10555, Marlin developpers are waiting an answer from host developpers.

GMagician, 2018-04-28:
@Zer0t3ch you can read all here: #10554

boelle, 2019-02-20:
@Zer0t3ch 

still having issues?

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
This is done and should be closed.

closed 2019-07-11 by shitcreek