MarlinFirmware/Marlin #10547: [FR] - change GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y by G29 command

MasterPIC, 2018-04-27:
Hi,

I'm using code from my PR #7923 to quickly set the size of the square leveling area (related to the size of the object to be printed). This way I don't waste time and I use the leveling probes only where required...
But when setting a bigger leveling area I'd like to increase GRID_MAX_POINTS_X and  GRID_MAX_POINTS_Y too.

What about making **G29 able to set GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y**?

Roxy-3D, 2018-04-27:
This isn't as easy as it appears.   The problem is the mesh based leveling systems have a two dimensional array  z_values[GRID_MAX_POINTS_X][GRID_MAX_POINTS_Y]  to hold the mesh points.

We could do this...  But that array would have to be declared at the maximum size.   And we don't want to do that because very few people want or need a 15 x 15 mesh.   Declaring it at that size would consume valuable RAM memory from other things for no purpose (in most cases).

thinkyhead, 2018-04-28:
Request appreciated, but it would be a pretty major rewrite.
We'll put it on the TODO list for Marlin 2.0.

qwewer0, 2021-01-01:
@MasterPIC Could UBL be the answer for the FR? For UBL we only need 3 point leveling (G29 J) at the start of a print.

MasterPIC, 2021-01-02:
@qwewer0 There isn't enough memory to enable UBL on my printer. Anyway I reverted to glass on the hotbed and so I don't need any leveling mesh.

KodeMunkie, 2021-08-02:
I raised a duplicate issue for this one #22489 and have a comment from that I'd like to add:
When regional bed levelling is used can we just disable the current ABL implementation ("Bed level off" in the UI) and just reuse that feature's grid array?
This way the behaviour flow is exactly the same as "Bed level on" but uses a smaller subset of the grid data, without the risk of inconsistently normalised _total_ grid of points that having the "Bed levelling feature" left on would cause.

KodeMunkie, 2021-08-03:
I'm assuming this is no longer a priority for anyone but my use case I don't believe is a "pretty major rewrite" so I'm happy to  do this work myself and it won't match this specific issue's requirements (specifically I won't be setting grid max points and such). As such I **won't** post an MR against this issue.

The flow of this is (pseudocode, only read this Marlin code the other day for first time, love that it's easy to follow :) ):

- If (G11111) // arbitrary choice will also have enable/disable flag
AND
- If (planner.leveling_active == false)
THEN
- Get bounding box of G-CODE (TBC - possibly a new gcode from slicer?) derive start_at_with_offset, end_at_with_offset.
- G29 S<start_at_with_offset> E<end_at_with_offset> // update in memory mesh with points used in the model only
- Continue BAU print following planner.leveling_active == true path (do not enable LCD menu option).
- Finish and run M420 L0 (Not checked this code - just restore the values from EEPROM).

Please let me know whether you'd like the work putting against another issue, want to redefine this issue or happy for me not to contribute.

thisiskeithb, 2023-07-15:
Closing since there’s now a PR to add support:

- #26084

closed 2023-07-15 by thisiskeithb