MarlinFirmware/Marlin #10550: [2.0.X] Compilation fails

th33xitus, 2018-04-27:
### Description
Everything was set up correctly but compilation of the firmware fails.
Error message tells me:
`#error "SENSORLESS_HOMING requires X_MIN_ENDSTOP_INVERTING when homing to X_MIN."`
But i have it inverted:
`#define X_MIN_ENDSTOP_INVERTING true`

#### Additional Information
Chosen board:
`#define MOTHERBOARD BOARD_RAMPS4DUE_EFB`

[image]

thinkyhead, 2018-04-28:
The error will only appear if `X_MIN_ENDSTOP_INVERTING` is `false`. It's possible that you've edited a different configuration file from the one in the root folder. Quit Arduino IDE and re-start it. Double-check that the correct file (`Marlin/Configuration.h`) has in fact been edited. Then re-build. The error should go away if those conditions are confirmed.

th33xitus, 2018-04-28:
@thinkyhead I can assure you, that the files i am changing are in the right folders.
How do i know? I made a change in the config, saved it, looked at the "last modified timestamp" in my explorer and it tells me the current time. So it is definitely the right config-files i am editing here.
Error doesn't go away.

**Here are my configuration files:**
[configuration.h_configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/1957770/configuration.h_configuration_adv.h.zip)
**Commit i used:**
a474a7e

Also i get another problem with compiling while having TMC2130 and SD_SUPPORT enabled both at a time. You want me to make a new bugreport for it or is it okay to discuss it here as well?
Having TMC2130 and SD_SUPPORT enables gives me the following error message while trying to compile:
`DUE software SPI is required but is incompatible with TMC2130 hardware SPI. Enable TMC_USE_SW_SPI to fix.`
Enabling software SPI is actually not an option since i want to use hardware SPI.
But i would love to use SD_SUPPORT as well.

thinkyhead, 2018-04-28:
Well that is interesting! This prints both error lines:
```cpp
#if X_MIN_ENDSTOP_INVERTING == true
  #error "X_MIN_ENDSTOP_INVERTING == true"
#endif
#if DISABLED(X_MIN_ENDSTOP_INVERTING)
  #error "X_MIN_ENDSTOP_INVERTING is also somehow 'DISABLED'!"
#endif
```
My guess at this point is that something in the DUE platform is messing with the DISABLED macro. Some more investigation to follow…

th33xitus, 2018-04-28:
Okay :stuck_out_tongue: 
I "bruteforced" the compilation with just deleting the corresponding sanity check for sensorless homing and then it compiled fine. But i get an instant driver error on startup. Didn't figure out the exact driver error i get since i wasn't able to get my OctoPrint connected to the Due (doesn't recognizes the USB-port for whatever reason) so i couldn't get the console output yet. 

But i guess it's better to wait anyways so you guys can take a look and might fix the stuff which is broken instead of me doing workarounds with disabling sanitychecks and such stuff :smiley:

thinkyhead, 2018-04-28:
The Due platform defines `false` as `0x0` and `true` as `0x1`, so this change has to be made to `macros.h`:
[image]

thinkyhead, 2018-04-28:
Now merged into `bugfix-2.0.x`.

closed 2018-04-28 by thinkyhead

thinkyhead, 2018-04-28:
> you want me to make a new one?

Not if the "`SD_SUPPORT` and TMC2130 HW SPI conflict" issue already has a topic.
Anyway, no, I'm not working on that issue. That's outside my purview.

StealthNinja1O1, 2019-06-27:
I downloaded the latest marlin, and still got that error. I commented it, and i'll report back when i finaly get the printer in. SKR V1.3 TMC2209 btw.
`  #if ENABLED(DELTA) && !BOTH(STEALTHCHOP_XY, STEALTHCHOP_Z)
    #error "SENSORLESS_HOMING on DELTA currently requires STEALTHCHOP_XY and STEALTHCHOP_Z."
  //#elif X_SENSORLESS && X_HOME_DIR == -1 && (X_MIN_ENDSTOP_INVERTING != X_ENDSTOP_INVERTING || DISABLED(ENDSTOPPULLUP_XMIN))
    //#error "SENSORLESS_HOMING requires X_MIN_ENDSTOP_INVERTING and ENDSTOPPULLUP_XMIN when homing to X_MIN."
  #elif X_SENSORLESS && X_HOME_DIR ==  1 && (X_MAX_ENDSTOP_INVERTING != X_ENDSTOP_INVERTING || DISABLED(ENDSTOPPULLUP_XMAX))
    #error "SENSORLESS_HOMING requires X_MAX_ENDSTOP_INVERTING and ENDSTOPPULLUP_XMAX when homing to X_MAX."
  //#elif Y_SENSORLESS && Y_HOME_DIR == -1 && (Y_MIN_ENDSTOP_INVERTING != Y_ENDSTOP_INVERTING || DISABLED(ENDSTOPPULLUP_YMIN))
    //#error "SENSORLESS_HOMING requires Y_MIN_ENDSTOP_INVERTING and ENDSTOPPULLUP_YMIN when homing to Y_MIN."
  #elif Y_SENSORLESS && Y_HOME_DIR ==  1 && (Y_MAX_ENDSTOP_INVERTING != Y_ENDSTOP_INVERTING || DISABLED(ENDSTOPPULLUP_YMAX))
    #error "SENSORLESS_HOMING requires Y_MAX_ENDSTOP_INVERTING and ENDSTOPPULLUP_YMAX when homing to Y_MAX."
  //#elif Z_SENSORLESS && Z_HOME_DIR == -1 && (Z_MIN_ENDSTOP_INVERTING != Z_ENDSTOP_INVERTING || DISABLED(ENDSTOPPULLUP_ZMIN))
    //#error "SENSORLESS_HOMING requires Z_MIN_ENDSTOP_INVERTING and ENDSTOPPULLUP_ZMIN when homing to Z_MIN."`