MarlinFirmware/Marlin #10551: [2.0.x] MKS SBASE disable MULTIPLE_PROBING did not work

haudamekki, 2018-04-27:
I'm using a Precision Piezo for probing my Z axis and want to disable MULTIPLE_PROBING by outcommenting it, but it did not work. After making the change and sending a G28 the Probe did  just the same, first a fast and then a slow probe. Also changin Z_PROBE_SPEED_SLOW to the same value as Z_PROBE_SPEED_FAST has no effect.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1956494/Configuration.zip)

Bob-the-Kuhn, 2018-04-27:
Try setting it to 1 rather than commenting it out.

thinkyhead, 2018-04-28:
```cpp
#if MULTIPLE_PROBING && MULTIPLE_PROBING < 2
  #error "MULTIPLE_PROBING must be >= 2."
#endif
```

thinkyhead, 2018-04-28:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

haudamekki, 2018-04-28:
I have tried it with the latest bugfix-2.0.x branch and have the same Problem.
Can't disable MULTIPLE_PROBING.
Can't change the speed for Z_PROBE_SPEED_SLOW
Can't set MULTIPLE_PROBING to 1 as mentioned above by thinkyhed

thinkyhead, 2018-04-28:
Please follow this procedure to gather more information:

- Disable `MULTIPLE_PROBING` so that only a single probe will be done.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

haudamekki, 2018-04-28:
I have disabled MULTIPLE_PROBING and enabled DEBUG_LEVELING_FEATURE.
Performing a G28 the Printer does a fast and a slow probe.
Performing a G29 the Printer does only one probe per Point.

[debug.txt](https://github.com/MarlinFirmware/Marlin/files/1957817/debug.txt)

haudamekki, 2018-04-28:
Sorry it was my own Fault...
It's not probing, it's Z_HOME_BUMP settings I'm searching for.
Can I disable the home bump for Z?

thinkyhead, 2018-04-28:
Yes, but it may also affect your `G29`. I've just added a patch to the bugfix branches so you can just set `Z_HOME_BUMP_MM` to 0 and it will only affect `G28`.

haudamekki, 2018-04-28:
THX works perfect