MarlinFirmware/Marlin #10564: STM32F4/F7 - how to compile?

xC0000005, 2018-04-28:
How are people building STM32F4 or F7 board firmware building? I can't get PlatformIO to use the generic STM32 core. I'd like to build support for an F070 board, which the generic core supports.

thinkyhead, 2018-04-29:
See #10434, https://github.com/danieleff/STM32GENERIC, and https://danieleff.github.io/STM32GENERIC/

boelle, 2019-02-20:
@xC0000005 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

xC0000005, 2019-05-19:
This can be closed.

closed 2019-05-19 by xC0000005