MarlinFirmware/Marlin #10567: [2.0.X] Compilation fails - UBL for RAMPS FD V2

kevintong116, 2018-04-29:
following error occur if ubl is enabled
Marlin\src\module/configuration_store.h:100:47: error: 'E2END' was not declared in this scope

E2END is not defined in pins_RAMPS_FD_V2.h

cjsoong, 2018-04-29:
@kevintong116 
#define BOARD_NAME         "RAMPS-FD v2"

#include "pins_RAMPS_FD_V1.h"

#undef INVERTED_HEATER_PINS
#undef INVERTED_BED_PINS
#undef INVERTED_FAN_PINS
#define E2END 0xffff  // 64Kb (24LC512) .. 0x7fff(32Kb 24LC256)
#define I2C_EEPROM

p3p, 2018-04-29:
@thinkyhead We need to add that constant into the PersistentStorage class as it will depend on the active implementation, Roxy started using it after I had written the abstraction. 

> Is there a specific size that we can actually depend on?

I think the minimum is 2k though I'm unsure, @Roxy-3D would have a better idea

thinkyhead, 2018-04-30:
@cjsoong seems to be pointing out that it depends on which daughter module you use. One has 64K and the other has 32K. Marlin really only needs 1K, but the extra space can’t hurt. I’d say we can reliably set it to 4K but only if E2END hasn’t been set elsewhere, such as in a PersistentStore class that uses the SD card.

Roxy-3D, 2018-04-30:
> I’d say we can reliably set it to 4K but only if E2END hasn’t been set elsewhere, such as in a PersistentStore class that uses the SD card.

Agreed.  4KB is a good default number.

boelle, 2019-02-20:
@kevintong116 still having issues with latest bugfix 2.0?

closed 2019-07-09 by shitcreek