MarlinFirmware/Marlin #10570: M303 PID autotune error 2.0x

cdedwards, 2018-04-29:
M303 S220 C25 
will autotune and provide values
M303 S221 C25
will autotune fail with Temperature too high.
 PID_FUNCTIONAL_RANGE  I changed from 10 to 25 which should have given me enough room to make this work, but the temperature overshoots considerably. This is consistant.

cdedwards, 2018-04-30:
isn't this rather specific though? S220 works. S221 does not. MAX_OVERSHOOT_PID_AUTOTUNE allows us to overshoot by 20 degrees. I'm hitting 240ish on both of those numbers and only failing when it's 221. 
I'm not close to the max temp of my hotend as I have a Prometheus. I should be able to hit 300-330 easily with it as there is no PTFE in the hotend.

thinkyhead, 2018-04-30:
The only code that throws that error:
```cpp
      if (current > target + MAX_OVERSHOOT_PID_AUTOTUNE) {
        SERIAL_PROTOCOLLNPGM(MSG_PID_TEMP_TOO_HIGH);
        break;
      }
```

cdedwards, 2018-04-30:
When I get my printer re-assembled, I'm going to try reducing the 5000ms down to about 3000ms. and see what that gains me. 

if (heating && current > target) {
   if (ELAPSED(ms, t2 + 5000UL)) {

if (!heating && current < target) {
    if (ELAPSED(ms, t1 + 5000UL)) {

[image]

thinkyhead, 2018-04-30:
No rush! Be sure to try setting the max overshoot to 100 and that should at least bypass the error.

closed 2018-06-04 by cdedwards