MarlinFirmware/Marlin #10573: Prusa Multiplexer - active_extruder should always be 0

mtm88001, 2018-04-30:
### Description

I have built a custom 3d printer using a ramps based board and the Prusa Multiplexer.  I installed the latest Marlin 1.1.8 and changed only enough of the config to enable the MK2_MULTIPLEXER.  Printer works; however, I am only able to extrude from extruder 0.  I was able to change code in Marlin_main.cpp to fix the issue and wanted to get feedback on the code change.

bugfix 1.1.x seems to have the same code.

### Steps to Reproduce

1. [First Step]
Using Repetier-Host with printer heated up and ready to print with four extruders configured.
2. [Second Step]
Change to Extruder 0 and try to extrude and the printer extrudes.
3. [Third Step]
Change to any Extruder other than Extruder 0 and try to extrude.  The printer will not extrude.

**Expected behavior:** [What you expect to happen]
When switching extruders from Repetier-Host the Multiplexer should click indicating a switch and should be able to extrude from the selected extruder.
**Actual behavior:** [What actually happens]
When switching extruders from Repetier-Host the Multiplexer does click indicating a switch and selecting any extruder except Extruder 0, the printer will not extrude.
#### Additional Information

Current code snippet:
Marlin_main.cpp 1.1.x
 #if ENABLED(MK2_MULTIPLEXER)
        if (tmp_extruder >= E_STEPPERS)
          return invalid_extruder_error(tmp_extruder);

        select_multiplexed_stepper(tmp_extruder);
      #endif

      // Set the new active extruder
      active_extruder = tmp_extruder;

AcroWD Fork
Marlin_main.cpp 1.1.8
#if ENABLED(MK2_MULTIPLEXER)
        if (tmp_extruder >= E_STEPPERS)
          return invalid_extruder_error(tmp_extruder);

        select_multiplexed_stepper(tmp_extruder);
        active_extruder = 0;
      #else
          // Set the new active extruder
          active_extruder = tmp_extruder;
      #endif

See https://github.com/mtm88001/Marlin/tree/AcroWD for reference.

Additionaly I found a code incosistency, not sure if on purpose or just got missed.  Should the for loop not reference s instead of e since s is the index?

Marlin_main.cpp 1.1.x
/**
   * M702: Unload all extruders
   */
  inline void gcode_M702() {
    for (uint8_t s = 0; s < E_STEPPERS; s++) {
      select_multiplexed_stepper(e);

  
AcroWD Fork
Marlin_main.cpp 1.1.8
 /**
   * M702: Unload all extruders
   */
  inline void gcode_M702() {
    for (uint8_t s = 0; s < E_STEPPERS; s++) {
      select_multiplexed_stepper(s);

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-04-30:
> I was able to change code in Marlin_main.cpp to fix the issue and wanted to get feedback on the code change.

Please actually test the `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we can investigate further.

> Additionaly I found a code incosistency

We don't patch 1.1.x. That branch is for tagged releases only. You should look at the current `bugfix-1.1.x` and `bugfix-2.0.x` branches to see where current development is. The M702 lines you're displaying are no longer present in the latest code.

revilor, 2018-05-09:
Sounds like the DISABLE_INACTIVE_EXTRUDER issue which is fixed in the bugfix branches. For 1.1.8

#define DISABLE_INACTIVE_EXTRUDER false

should resolve your issue.

closed 2018-05-10 by thinkyhead