MarlinFirmware/Marlin #10574: Q: What happens at filament runout with this config?

boelle, 2018-04-30:
tried on the facebook group but could not get a direct answer

My main concern is that i want to use the P2 parameter for nozzle park so that it does not go down to Z20 and hit a print but rather go up 20mm 

/**
 * Nozzle Park
 *
 * Park the nozzle at the given XYZ position on idle or G27.
 *
 * The "P" parameter controls the action applied to the Z axis:
 *
 *    P0  (Default) If Z is below park Z raise the nozzle.
 *    P1  Raise the nozzle always to Z-park height.
 *    P2  Raise the nozzle by Z-park amount, limited to Z_MAX_POS.
 */

Then secondary how do i test the status of the switch? M119 did not show the status

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1960185/Configuration.zip)

thinkyhead, 2018-04-30:
If you had tested the feature, that would have given you a very direct answer.

When pausing the print via `M125` or `M600` it always uses the P2 behavior, raising the nozzle by the Z value.

Since it would be foolish to lower the nozzle when pausing in the middle of a print, we coded it not to do that. And if we had coded it to do that, someone would have slapped us hard and gotten us to fix it very quickly.

thinkyhead, 2018-04-30:
> Then secondary how do i test the status of the switch? M119 did not show the status.

You have to enable `FILAMENT_RUNOUT_SENSOR`.

boelle, 2018-04-30:
> When pausing the print via M125 or M600 it always uses the P2 behavior, raising the nozzle by the Z value.

Just had to be sure, but yes i could imagine you could get  5 red fingers in the face otherwise

it was just the parameter list that made me shake my hand :-)

> You have to enable FILAMENT_RUNOUT_SENSOR.

DOH!

reflashing and will close if everything works as expected

boelle, 2018-04-30:
#error "FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH."

does not quite make sense to me as a filament unload is not extrude and is kind of special operation

but for now i will just set max extrude to 1000mm

boelle, 2018-05-01:
printed wrong part so filament switch will not be on before later today

can you tell what happens if i ask for a filament change on the lcd while not printing, will it heat up first?

i might get to it before you can give in an reply

boelle, 2018-05-02:
@thinkyhead i wanted to close this one myself unless you protest

and please could you tell me what happens if i from lcd ask for a filament change and the hotend is stone cold?

boelle, 2018-05-02:
i tried to do an unload and it does ask for what temp to preheat to

i just did not get the 900mm i expected as i have #define FILAMENT_CHANGE_UNLOAD_LENGTH      900

maybe its this one for an unload only? #define FILAMENT_UNLOAD_RETRACT_LENGTH

i skipped that one as the texty says  // Filament Unload does a Retract, Delay, and Purge first:

but another surprise is that it did not turn off the nozzle

i would do an unload if the printer is not in use for an extended period and i want to store the filament in dry box

boelle, 2018-05-02:
so i redid the unload and it did retract wait and then it loaded a bit again

and it lowered Z about 20mm

since this it from non printing i would only expect it to unload, but i have changed these lines

#define FILAMENT_UNLOAD_RETRACT_LENGTH     900
#define FILAMENT_UNLOAD_DELAY                         0
#define FILAMENT_UNLOAD_PURGE_LENGTH          0

reflashing and testing if that works as the "unload" word says

next would be to get the load part to work but i will assume that its just an unload in the oposite direction

closed 2018-05-02 by boelle

thinkyhead, 2018-05-03:
You don't want the `FILAMENT_UNLOAD_RETRACT_LENGTH` of 900. That setting is for the small initial retract before unloading… Set it to a small value like 5 or leave it set to 0 if you don't care about the small retract/delay/purge move.
```cpp
// Filament Unload does a Retract, Delay, and Purge first:
```

Set `FILAMENT_CHANGE_UNLOAD_LENGTH` to 900. See the description:
```cpp
#define FILAMENT_CHANGE_UNLOAD_LENGTH 100  // (mm) The length of filament for a complete unload.
                                           //   For Bowden, the full length of the tube and nozzle.
                                           //   For direct drive, the full length of the nozzle.
                                           //   Set to 0 for manual unloading.
```