MarlinFirmware/Marlin #10577: Execution of G29 (ABL Bilinear) command really slow.

mrah, 2018-04-30:
### Description

Hello, today i flashed my firmware and installed **Marlin-bugfix-1.1.x**. But i have problem with ABL feature. G29 command process working slow. There is too much delay between probing. You can see in video. I have no problem with Marlin 1.1.x.

#### Additional Information
My configuration files.
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/1960597/Archive.zip)

Video
Part: 1
https://youtu.be/VA7VZRxYn38
Part: 2
https://youtu.be/FF_54ccANiM

Zwaubel, 2018-04-30:
I can confirm the same behavior on my machine and I'm pretty sure this problem was introduced by the commit #4dfc011d of @thinkyhead. The problem probably is caused by a double conversion of the slow z feed rate from mmm to mms somewhere in the code.

mylife4aiurr, 2018-04-30:
Upgraded this morning to commit 071387bace113c10be90f8b0a99306bbbc7d591d. Delta machine.

Homing G28 works fine.
Movements in Octoprint work fine. ie home z, move any direction 100mm works fine.

However G33 autodeltaconfig. G33 P3 V2 F6
Starts at z max. Starts moving toward bed fast, then switch to super slow speed to probe the bed. I end up sending M112 because its so slow.

This happened between
commit b4ddee8bebead400142fe45edc3bc41bea4ecb16
commit 071387bace113c10be90f8b0a99306bbbc7d591d

Zwaubel, 2018-04-30:
Checkout @MagoKimbra PR: #10576
This addresses the super slow homing issue and fixed the problem for me.

mylife4aiurr, 2018-04-30:
@Zwaubel confirmed fixed with the commit you referenced.

thinkyhead, 2018-04-30:
Thanks! Feedrates for both homing and probing moves should now be consistent.

closed 2018-04-30 by thinkyhead