MarlinFirmware/Marlin #10578: Option for minimum LCD support

Looxoor, 2018-04-30:
### Description

Since I have a board with very constrained program memory (Anet) and solely control the printer via Octoprint, I would like to be able to strip the firmware of all menus, essentially just use the LCD to show the normal status screen.
In my current config I'm using 98% of the available memory and there are several features I'd like to enable (without purchasing a new board).

I'd like an option in e.g. configuration_adv.h:

```
// Disables all menus and use LCD for status screen only.
// This is useful if you need to free up program memory 
// and only use an external controller for your printer
#define LCD_STATUS_ONLY
```
I believe this would benefit more people than me.
Although I understand that this might be too much work if I'm a minority with these wishes.

MasterPIC, 2018-04-30:
Did you define SLIM_LCD_MENUS (in configuration_adv.h)?

Looxoor, 2018-04-30:
All right, I'll definitely try that and see if I can enable G26, linear advance and advanced pause.
Thanks!

Just curious though, did my idea at least make a little sense?

MasterPIC, 2018-04-30:
Surely some memory is still used for LCD, also with SLIM_LCD_MENUS defined...
I don't remember if I've mesh validation enabled: I simply make a G29 calibration with ABL before each print.
What I know is that I wasn't able to switch from ABL to UBL because of memory. But I don't think that disabling SD and LCD support would solve this limitation.

Looxoor, 2018-04-30:
Yeah, I think maybe not the individual menu items as much as the menu logic itself, i.e. The button actions and such would make the biggest impact to get rid of. Just off the top of my head.

Oh, I have ABL with bilinear grid as well, I'd just like a simple way to test the entire bed since I can't probe all the way to the front in my case.

Ah, I'm still on mainline 1.1.8, is the bug fix branch stable enough to use full time? Reading about the differences I have opted out of even trying linear advance 1.0, waiting and hoping for 1.1.9 stable to come soon. Maybe I should just take the plunge then?

When I couldn't fit advanced pause I took the octoprint path of M0 instead of M600 for filament change procedure and it actually works flawlessly. I'd just prefer a native solution.

Looxoor, 2018-04-30:
Wow, that was fast, thanks @thinkyhead!

Looxoor, 2018-05-16:
Tried to compile it (downloaded the latest bugfix an hour ago) with the "NO_LCD_MENUS" option for the first time today and got two errors during compilation:

First, in "G26_Mesh_Validation_Tool.cpp":
```
'is_lcd_clicked' was not declared in this scope
```

I commented out the G26 option in configuration.h and tried to compile again,
then the next compile time error occured:

This time in temperature.cpp
```
'ADC_KEYPAD_PIN' was not declared in this scope
```

So I had to comment out no lcd menus for now and use slim lcd menus instead.

Hope this helps.

thinkyhead, 2018-05-19:
@Looxoor — These errors should be cleared up now.

Transvistinator, 2018-11-06:
Hate to re-open an old issue, but getting the same error plus a bunch more when enabling NO_LCD_MENUS.

Slim menus works, but Isn't freeing up enough :/

https://pastebin.com/QKBL09N9