MarlinFirmware/Marlin #10579: [bugfix-2.0.x] Amplified feedrates on delta kinematics

vilsed, 2018-04-30:
Hello, everyone.

After compiling the latest 2.0.x for my custom delta with Re-Arm, came across a bizarre issue. Feedrates of movements controlled from the LCD menus are fine (i.e. "move axis", homing, etc.), but printing feedrates seem to be 10x higher than they should be. It doesn't matter if I'm printing from SD or host, same same deal. Although while printing from the SD card, printer stops for a second quite often, as if the buffer becomes emptied.

Are there higher forces involved or am I missing something?

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1961205/Marlin.zip)

vilsed, 2018-05-02:
I'm thinking it might be somehow connected with #8778, since Marlin from Jan 17 works properly in this regard. The maths and code behind that PR is way beyond my paycheck, so it's more of a guess. @thinkyhead @tcm0116 what do you think?

vilsed, 2018-05-05:
Was digging through the code these last few days. Didn't find anything concrete, but found that current 2.0.x branch motion.cpp, motion.h, planner.cpp and planner.h are different from those in the #8778 PR, which was merged back then as I understand. So I'm completely lost in the woods here. And the weirdest thing is that I don't see any other delta users having this problem.

vilsed, 2018-05-19:
Hello. At some point, someone fixed this issue. Everything seems to be working fine now. Closing.

closed 2018-05-19 by vilsed

thinkyhead, 2018-05-20:
Sorry I missed this issue post. I must have been out to lunch. I'm glad it was fixed!