MarlinFirmware/Marlin #10583: Einsy Rambo 1.1 issues with Reprap discount smart controller

danman454, 2018-05-01:
Hey all. i'm pretty new to all this but have always been able to figure any issues out by myself with web searches. I think ive bitten off more than i can chew with this one though.

I recently purchased the Einsy Rambo v1.1 board from 3dMakerWorld as an upgrade for one of my printers. I also bought a new reprap discount smart controller from Amazon. I cannot for the life of me get the display to work. At most I get a blue screen with white squares. Ultamachine was able to prove the board and display work properly by having me upload Prusa firmware for MK3 via their firmwareupdater v2.
But i cannot upload any firmware through arduino that makes the display function like it does using Prusa mk3 hex file.

I've tried several versions of Marlin and have been web searching for about a week. i'm at my wits end. Any help will be appreciated.

danman454, 2018-05-01:
Sorry. These are from the bugfix1.1.x

[einsy_rambo.zip](https://github.com/MarlinFirmware/Marlin/files/1965486/einsy_rambo.zip)

kAdonis, 2018-05-02:
After a quick look, I see you have defined the wrong motherboard in Configuration.h 
Try:

`  #define MOTHERBOARD BOARD_EINSY_RAMBO`

danman454, 2018-05-02:
Ok so i changed the board type like you suggested then i get this error: 

In file included from sketch\stepper.h:48:0,

                 from sketch\Marlin_main.cpp:256:

sketch\stepper_indirection.h:57:30: fatal error: TMC2130Stepper.h: No such file or directory

   #include <TMC2130Stepper.h>

                              ^

compilation terminated.

exit status 1
Error compiling for board RAMBo.

danman454, 2018-05-02:
Fixed that error. screen works now. guess all i have to say now is THANK YOU! :)

closed 2018-05-03 by thinkyhead