MarlinFirmware/Marlin #10586: upgrade to 1.1.8 and lost heat bed display with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

fbmfbm, 2018-05-01:
Hi all !

After upgrading from 1.1.5 to 1.1.8 i'm loosing the heat bed display infos on LCD with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER .
I'm testing with 1.1.* buxfix but don't find any way to resolve that.

After reading lot of ressources i'm finding that some peoples can have it to displaying by allowing  THERMAL_PROTECTION_BED but this option is ON by default.

it's a bug or someone have a solution ?

Thank's.

thinkyhead, 2018-05-01:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

fbmfbm, 2018-05-01:
Thank's for your answer thinkyhead,

Well ;#....;my apologizes.... My TEMP_SENSOR_BED is at 0. Don't think o check that as it's ok by default in prevues versions.
Ok you can close this.
by

closed 2018-05-01 by fbmfbm