MarlinFirmware/Marlin #10591: Ramming walls with UBL

evg-zhabotinsky, 2018-05-01:
I've had a weird problem yesterday. While printing bottom layer infill near the edge of the table, print head didn't stop after completing yet another line. It continued in the same direction with the same speed and eventually rammed the wall, while printer was acting like it still didn't finish that last segment.

I've slimmed gcode down to a few movement instructions while still reproducing this, and it only happened with leveling enabled and below leveling fade Z.

That was on 9769d799df76c1df0cfbeb78c9da02219f8a689c with UBL enabled, and it never happened to me in prior weeks. It also no longer happens on recent bugfix-1.1.x, but that might be a result of shuffling bits around and randomness.

The question is, is it worth it to track down what really happened? If yes, I don't really know where to start. The code base is kinda huge and you cannot even press ^C when it goes off rails to see the stack trace.

p3p, 2018-05-01:
I'm not sure but your bad build was 18 days ago, https://github.com/MarlinFirmware/Marlin/commit/c91b3142aaf6900396022ea83d6a212573534d6d (from 12 days ago) fixed a problem introduced 19 days ago, so hopefully this is resolved

Roxy-3D, 2018-05-01:
> I'm not sure but your bad build was 18 days ago, c91b314 (from 12 days ago) fixed a problem introduced 19 days ago, so hopefully this is resolved

And specifically...   That line of code that was reverted had to do with moves that crossed mesh cell boundaries where the calculations say to keep going longer than they should due to floating point rounding errors.

closed 2018-05-06 by evg-zhabotinsky