MarlinFirmware/Marlin #10597: Does this work for all AVR processors?  Should we make some GPIO macro's and change things?

Roxy-3D, 2018-05-02:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

p3p, 2018-05-02:
> And digitalWrite()'s are fast, right?

No digitalWrite is hopelessly slow, no problem if your just turning something on or off every now and then but you should never generate a signal with it, digitalWrite is about 5us to toggle, direct port manipulation (with a compile time known pin) about 125ns

thinkyhead, 2018-05-02:
The only place where we still use `digitalWrite` or `digitalRead` is when the pin number is in a variable, so can't be handled by the pre-compiler. For all cases where we known what the pin number will be ahead of time, we use the fastio `WRITE` / `READ` macros.