MarlinFirmware/Marlin #10599: [bugfix-2.0.x] Unable to use pin D11 as E0_AUTO_FAN_PIN

benlye, 2018-05-02:
Recently something has changed which seems to be preventing D11 being used as `E0_AUTO_FAN_PIN` on a RAMPS board.

I have a RAMPS setup with a [fan extender](https://www.banggood.com/Geeetech-Reprap-Ramps1_4-RRD-Fan-Extender-For-3D-Printer-Reprap-With-2-Channels-p-1214050.html) providing 2x switched 12V supply pins, switched by pins D6 and D11.  

I have this in Configuration_adv.h:
```
#define E0_AUTO_FAN_PIN 11
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed
```

And I'd been happily using D11 as the fan pin until I updated my firmware to the latest bugfix-2.0.x yesterday.

Now, when the extruder hits 50 degrees the LED on the fan extender board lights up to signal that it has received a signal, but 12V does not flow through to the fan.

After measuring everything to confirm that the fan works and 12V is getting to the input side of the fan extender I used M43 to debug the state of D11:

```
Recv: PIN:  11   Port: B5        E0_AUTO_FAN_PIN                        Output = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled
Recv: .                          SERVO0_PIN                             Output = 0
```

When the extruder is hot the pin switches to `Output = 1`.  What struck me as odd is the `non-standard PWM mode` flag and the PWM value of 2000.

I noticed that D6 (the other pin I can use on the fan extender) did not have the PWM flag:
```
Recv: PIN:   6   Port: H3        SERVO1_PIN                             Input  = 0    TIMER4A   PWM:     0    WGM: 1    COM4A: 0    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
```

So I switched my `E0_AUTO_FAN_PIN` to D6, moved the fan connection to the corresponding pin, and now the fan spins as expected when the extruder is hot.

My question is why the change to D11?  Is this expected?  I imagine that others who are using D11 and a fan extender might be scratching their heads as well.

thinkyhead, 2018-05-02:
But it works ok with `bugfix-1.1.x`?

I believe that Timer 1 is used with OCR1A in mode 4 (WGM_CTC_OCRnA) to drive the stepper ISR. I'm not sure what this means for the behavior of Pin 11, except that it can't be used for PWM, only for basic digital IO.

benlye, 2018-05-02:
> But it works ok with bugfix-1.1.x?

I don't know about bugfix-1.1.x, but I'll try to test that.  Until recently it was working in bugfix-2.0.x so something has changed there relatively recently.

benlye, 2018-05-04:
Tested bugfix-1.1.x and the behaviour (and output from M43) is the same.

J-charles-C, 2018-05-21:
I,

Test #define E0_AUTO_FAN_PIN   P1_20 

best

boelle, 2019-02-20:
@benlye is this still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead