MarlinFirmware/Marlin #10604: [bugfix-2.0.x] Filament unload hangs the printer, lcd unresponsive

boelle, 2018-05-02:
So in https://github.com/MarlinFirmware/Marlin/issues/10574 i started to experiment with the options in the lcd and for unload i have now set it to

#define FILAMENT_UNLOAD_RETRACT_LENGTH 900
#define FILAMENT_UNLOAD_DELAY 0
#define FILAMENT_UNLOAD_PURGE_LENGTH 0

as i wanted what the word say just an unload and not a purge after it

but the printer froze up and i had to reset it, will try again and see how much i managed to unload before it happens and then update this post

thinkyhead, 2018-05-03:
**You don't want a `FILAMENT_UNLOAD_RETRACT_LENGTH` of 900.** That setting is for the **small initial** retract before unloading… Set it to a small value like 5 or leave it set to 0 if you don't care about the small retract/delay/purge move.
```cpp
// Filament Unload does a Retract, Delay, and Purge first:
```

**Do set `FILAMENT_CHANGE_UNLOAD_LENGTH` to 900.** See the description:
```cpp
#define FILAMENT_CHANGE_UNLOAD_LENGTH 100  // (mm) The length of filament for a complete unload.
                                           //   For Bowden, the full length of the tube and nozzle.
                                           //   For direct drive, the full length of the nozzle.
                                           //   Set to 0 for manual unloading.
```

boelle, 2018-05-03:
morning, will do at once

if printer is stone cold and i just do the unload can i expect that it turns of the heater once the unload is done?

boelle, 2018-05-03:
i must be very stupid
when asking for an unload from stone cold now it  just heats up and lowers Z by 20mm
no retract/unload and it does not turn off heater

the ADVANCED_PAUSE_FEATURE section as is: https://pastebin.com/RJXynqB3

Roxy-3D, 2018-05-03:
We need some careful testing of the advanced pause and filament change capability.   Those two features got combined and there have been a number of people having problems with them.   It maybe that the comments are clear enough to get the options set up correctly.   It is also possible that the complexity of combining those features caused some bugs to be introduced.

Either way...    We need to understand where the issues are and get them put to bed.

boelle, 2018-05-10:
@thinkyhead & @Roxy-3D 

am i misunderstanding the whole thing? or have i just gotten the config wrong compared to what i want to achieve ?

Roxy-3D, 2018-05-10:
I don't know the answer.   I think you will have to experiment.   But I can make a few comments that might help you...

```
#define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.
#define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
```
> to me that means that if it wants to park for some reason it will retract 2mm at 60mm/s. But for me it also implies that its currently printing. why would it park if its not printing and stone cold?

I don't believe printing or not matter with these settings.  (I would have to dig through the code to know for sure.)   But I can tell you the intent.   If the nozzle is hot,  and perhaps in the middle of a print, this does an initial retract so hot filament does not keep ooozing out of the nozzle.

Then...  (it is my belief) the nozzle is moved to the 'Park' location.    And next, the specified full retraction is done.   

> why would it park if its not printing and stone cold?

Because it is possible the extruder needs to be in a specified position to change the filament.   And in most people's case...   doing it well away from a partially printed part just makes sense.     And remember, the current embodiment is a combined Pause & Filament Change.    If the printer was busy and then was told to Pause, is it printing???   My belief is it doesn't care if it is actually printing or not.   It will do these operations to Pause or Change Filament.   (I could be wrong about that...    I just use it for filament change reasons when it runs out of filament.)

> This i understand as a unload, can be either while printing or if i just ask for an unload from cold. if not printing it should turn of heaters while done. if not printing Z movement should not happen.

It is my belief that the hotend will turn off if it is not told to continue quickly.   And it is 18 months ago, but we added some beeps so the user would know the printer was waiting for something to happen.   Even if the nozzle turns off...  When the user clicks the encoder wheel, it will start heating the nozzle to resume the filament change operation.

```
  #define PAUSE_PARK_NOZZLE_TIMEOUT 45        // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS 5 // Number of alert beeps before printer goes quiet
```

boelle, 2018-05-10:
well i have tried to set it up as logical as i could, the park part of it works but it does not do the unload and nor does is it do the load.. only the purge

so unless someone can give me the excat config i need to make it do as i want i will have to disable things again

i can live with neading to have some scripts on sd to do an load and unload when not printing, but it i cnt get it to do the unload at filament runout and make a load before the purge there is not much sense in using it at all

Roxy-3D, 2018-05-10:
You are mostly trying to use it  for the case when the filament runs out during the middle of a print, right?

If so...   Grab the Pause and Filament Change settings in the gMax_1.5+ configuration_adv.h file.   Those settings are how my printer is configured and they work.  (At least they did the last time I updated the firmware).    And once you verify they work...  You can tweak the numbers for how much to back out the filament and such...     

My point is...  start with something that works and then migrate to your settings.

boelle, 2018-05-10:
hehehe, problem was i did not have anything to start with :-D

but now i have, and yes it was during print mainly. being able to change and unload without printing would just be a bonus, but for that i can have a few scripts on sd or on octoprint

boelle, 2018-05-10:
so if i understand your settings right at runout it will retract 2mm and then park and back out 100mm right?

and then ask for new filament? but how does it know how long to load once you have inserted the filament?

thinkyhead, 2018-05-10:
> completely missunderstand the config…?
```cpp
#define ADVANCED_PAUSE_PURGE_LENGTH 100
```
Do you really want to purge 100mm of filament on every load? It seems like an awful waste of material, and must add up to quite a pile.

```cpp
#define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.
```
60mm/s extrude rate is crazy fast. At 838 steps/mm of a typical extruder, that multiplies to 50,280 steps per second. Your E max feedrate is 25mm/s, so it should limit the speed to 20,950 steps per second. Still, depending on other factors, the AVR processor may blow up (reboot) given such a high rate.

A good rule of thumb is to start with modest settings and raise them gradually until you find a good balance between speed and quality.

Roxy-3D, 2018-05-10:
The load length isn't that important (for me!!!!).   The reason is, I put the new filament in and push it into the extruder until I see something come out the nozzle.    At this point the LCD Panel lets me purge a small amount (again and again) until I'm ready to continue.    Mostly...  The only time I continually purge (or prime) is if I'm doing a color change.    Some times, when going from Black to White...  It takes 5 or 10 purges to get the color clean.

boelle, 2018-05-10:
Hehe, with 800'ish mm Bowden tube and titan extruder it's another thing. My
thumb would start to hate me

But I guess that once I got it working for unload I could close this one
and put up a feature request for it to do load after I click the knob

>

boelle, 2018-05-10:
I could have done the same but I have arround 200mm tube before the
extruder so filament have a smooth path to the extruder from the spool.

Just pushing the filament would not be easy.  Need to think about it

boelle, 2018-06-03:
i had some time to experiment

```
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
                                                  // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE     10  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL        25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      900  // (mm) The length of filament for a complete unload.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
                                                  //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   6  // (mm/s) Slow move when starting load.
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH     0  // (mm) Slow length, to allow time to insert material.
                                                  // 0 to disable start loading and skip to fast load only
  #define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE   6  // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_FAST_LOAD_ACCEL     25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH   800  // (mm) Load length of filament, from extruder gear to nozzle.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
  //#define ADVANCED_PAUSE_CONTINUOUS_PURGE       // Purge continuously up to the purge length until interrupted.
  #define ADVANCED_PAUSE_PURGE_FEEDRATE        3  // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_PURGE_LENGTH         50  // (mm) Length to extrude after loading.
                                                  //   Set to 0 for manual extrusion.
                                                  //   Filament can be extruded repeatedly from the Filament Change menu
                                                  //   until extrusion is consistent, and to purge old filament.

                                                  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_RETRACT_LENGTH      13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY             5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.

  #define PAUSE_PARK_NOZZLE_TIMEOUT           45  // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS         10  // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.

  #define PARK_HEAD_ON_PAUSE                    // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE           // Ensure homing has been completed prior to parking for filament change
```

That config only unloads about 80-90'ish mm, it was hard to make a mark on the filament and at the same time trigger the switch.

the purge more on the display was arround the 50mm mark so that works. its only the 800 and 900 mm numbers that do not work

boelle, 2018-06-04:
would it make sense to close this one and repost the last entry as a new issue? or just change the title?

thinkyhead, 2018-06-10:
Does `M502`, `M500` help?

boelle, 2018-06-10:
i'm going to test it again tomorrow. just got home from a birthday and is wated

boelle, 2018-06-11:
i guess its has been a combination of M502, M500 when i made the last changes to firmware

I wanted to get my First layer height in the center of the bed perfect and for that i turned off UBL by doing M502, M500 and M501 

and then i just tested the runout and it worked like expected

closed 2018-06-11 by boelle