MarlinFirmware/Marlin #10606: PlatformIO & teensy20 boards - does it work?

Bob-the-Kuhn, 2018-05-02:
I was playing with PlatformIO & tried to download my TEENSYLU.  To my surprise the Teensy download ICON popped up and then immediately disappeared.  No chance to actually do the download.

This might be because I had managed to blow away the original teensy bootloader in my card and had to flash it with another.

Could someone try to upload their board via PlatformIO?

5DPRINT
BRAINWAVE
BRAINWAVE_PRO
PRINTRBOARD
PRINTRBOARD_REVF
SAV_MKI
TEENSY2
TEENSYLU

----

FYI - there's a new method that allows you to upload your board **using only 2 or 3 mouse clicks**.  It's documented in the latest [docs WIKI](http://marlinfw.org/docs/basics/install_platformio.html).

fiveangle, 2018-05-09:
PlatformIO does not know about the boot loader on the specific Teensy20pp boards, so unless you have an board made with an actual Teensy2.0++ device from Paul (none that I am aware of) uploading using PIO will never work on any of them.

Correct method is to compile then flash the hex with the appropriate tool for your boot loader.  For Printerboard RevF (DFU boot loader by default from Printrbot Co.) connected to Raspberry Pi for example:

1. jumper the boot jumper
2. punch reset button on board
3. lsusb (confirm connected board appears as DFU device)
4. sudo dfu-programmer at90usb1286 erase
5. sudo dfu-programmer at90usb1286 flash Marlin.hex
6. remove boot jumper
7. depress reset button on board
8. drink 🍺 

-=dave

Bob-the-Kuhn, 2018-05-09:
I'm getting an education.  I had **assumed** that all the AT90USB boards used the Teensy download tool.  

My goal is to automate the PlatformIO build & download process for the mentioned boards.  Any info on the correct download tool and/or  PlatformIO environment would be much appreciated.

Bob-the-Kuhn, 2018-05-10:
Do all the boards come with a DFU bootloader?

I have an HID bootloader on my Teensylu.  Can you point me to a useable DFU bootloader I can flash to my board?

Once I get my board updated I can start to play and have some confidence it applies to this class in general.

Bob-the-Kuhn, 2018-05-10:
> Let me know if you run into any craziness…

I resemble that remark!  🙂🙂🙂🙂

Some progress.

I was able to download the At90USB1286 DFU bootloader from the Microchip website and burn it to my Teensylu.  I was then able to program it via the Linux commands that Dave provided.  I was not able to program it via PlatformIO.

The Brainwave_PRO that I have was NOT found by the dfu-programmer.

As best I can tell PlatformIO can't program any of the AT90USB boards.  The Teensy20 environment is the only one that officially supports the AT90USB but it won't download.  

The standard Atmel AVR platform in PlatformIO will compile the AT90USB1286 but it's not on any of the 500 supported boards so there's not a nice way of programming them.

----

CDC bootloader
- BRAINWAVE_PRO
- SAV_MKI
- TEENSYLU

DFU/TeensyDuino bootloader
- TEENSY2.0++
- Printerboard (except possibly for some early ones)
- PRINTRBOARD_REVF

Unknown bootloader
- 5DPRINT
- BRAINWAVE

Bob-the-Kuhn, 2018-05-13:
I have PlatformIO generating images for the Brainwave_PRO board.  

If I upload them via the USBtinyISP and AVRDUDUE they run.

If I upload them via the bootloader and AVRDUDUE they **don't** run.  Dumping the memory and the fuses (again via AVRDUDE) shows that everything is the same.

ideas?

thinkyhead, 2018-05-13:
@whosawhatsis — Any idea about this??

whosawhatsis, 2018-05-13:
@thinkyhead I've never used PlatformIO. My boards have a CDC bootloader, and I use a version of the Teensy board definition that is slightly modified (just enough to customize the device identifiers).

Last time I tried, something in Marlin was failing to compile for the AT90USB1286. I've been meaning to see if it's fixed, but applying the necessary modifications to a fresh pull takes time I haven't had to spare recently.

Bob-the-Kuhn, 2018-05-14:
Sounds like you create the image using the Teensy support in the Arduino IDE.

How do you upload the image?

I'm using AVRDUDE and the AVR109 protocol to do the upload.

fiveangle, 2018-05-14:
> The Brainwave_PRO that I have was NOT found by the dfu-programmer.

Which is expected, as it uses the CDC bootloader (which you already realized :smile:)

> Unknown bootloader
> 
> 5DPRINT

As I mentioned above, according to 5DPrint themselves, they use the DFU bootloader: https://bitbucket.org/makible/5dprint-firmware/wiki/Flash%20Instructions

> I'm using AVRDUDE and the AVR109 protocol to do the upload.

That should work [according to the LUFA CDC documentation](https://w0.dk/~chlor/lufa/Bootloaders/CDC/Documentation/html/): 

> To load a new HEX file with AVRDude, specify "AVR109" as the programmer, with the allocated COM port. On Windows platforms this will be a COMx port name:
> 
> avrdude -c AVR109 -p at90usb1287 -P COM0 -U flash:w:Mouse.hex
> 
> On Linux systems, this will typically be a /dev/ttyACMx port name:
> 
> avrdude -c AVR109 -p at90usb1287 -P /dev/ttyACM0 -U flash:w:Mouse.hex
>
> Refer to the AVRDude project documentation for additional usage instructions.

-=dave

fiveangle, 2018-05-14:
Hey Bob, if you're in their compiling for BRAINWAVE_PRO anyway, maybe do a quick test on the kossel pro example config ?  It wasn't compiling when I was going through all the configs, and not sure if it was ever resolved:   https://github.com/MarlinFirmware/Marlin/issues/8795

-=dave

Bob-the-Kuhn, 2018-05-16:
I was able to compile the example configuration by changing the pins_BRAINWAVE_PRO.h file.

The problem was that the file only had one Z endstop (Z_STOP_PIN) but the configuration used Z_MIN for the Z_MIN_PROBE and Z_MAX for homing.  I just selected Z_STOP and added Z_MIN and Z_MAX pins.

Bob-the-Kuhn, 2018-05-25:
Finally able to download to the CDC bootloader cards.

I had to go back to an older version of AVRDUDE.

The latest one was getting confused at the 32K boundaries and leaving gaps in FLASH.

closed 2019-01-05 by Bob-the-Kuhn