MarlinFirmware/Marlin #10607: Bed has a tilt after G33

avri-schneider, 2018-05-03:
### Description

I have an Anycubic Kossel Linear Plus. It has a FIX_MOUNTED_PROBE (magnetic probe attached to effector with  a micro-switch directly below the nozzle).

It has a thick flat aluminium heated bed.

I am trying to use the latest Marlin firmware - it printed fine with hex file supplied by vendor, but it is an outdated Marlin version.

I am using the latest bugfix-1.1.x version pulled today.

I have G29 specifically disabled, to try and isolate the issue, as [apparently](http://hennerley.nz/2018/01/29/g29-vs-g33/) with a flat bed one should not need G29 on a Delta and should use G33 instead.

Attached Configuration.h and Configuration_adv.h:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1969290/Marlin.zip)

When trying to print, I notice the nozzle is too close to bed at one end and too high on another, indicating a tilted bed.

The printing was done following a 'G33 A' routine that was stored to EEPROM using 'M500'.

### Steps to Reproduce

1. Flash latest marlin firmware from branch bugfix-1.1.x, using supplied Configuration.h and Configuration_adv.h
2. M502
3. M500
4. Preheat PLA (wait for bed, hotend to reach printing temperature)
5. Attach magnetic probe to effector and connect probe's wire
6. G33 P1
7. M500
8. G33 A
9. M500
10. Disconnect probe's wire and detach magnetic probe from effector
11. Probe Z offset with paper - use LCD to adjust offset until nozzle is paper distance from bed
12. M500
13. Print bed level object and observe first layer

**Expected behavior:**
First layer should print evenly on the bed surface.

**Actual behavior:**
One side of the model is printed too high, and the opposite too low - indicating a tilt. 'G33 A' should have resolved the tilt.

thinkyhead, 2018-05-03:
> 'G33 A' should have resolved the tilt.

Only if the machine is 100% mechanically true. Even adjusting the endstop offsets can only do so much. @LVD-AC may have some additional advice on how to get everything perfectly flat. If you can operate without `G29` that will be best, but it can't hurt to use `G29` to correct for imperfections in the print surface.

> I notice the nozzle is too close to bed at one end and too high on another, indicating a tilted bed.

This can also indicate imperfections in the angle of the probe. Of course, if your nozzle is the probe then you can disregard this comment.

avri-schneider, 2018-05-03:
> Only if the machine is 100% mechanically true. 
A low SD of ~0.02 is not an indication of a good mechanical setup?

LVD-AC, 2018-05-03:
Then A parameter is no longer supported in the latest bugfix. The "calibration factors" (used internally) are now calculated internally with some smart use of forward and revers kinematics. Use G33 in the normal way.

Z-offset still has some issues in bugfix on deltas. You should set it before using G33 (or reissue a G33 command after each z-offset change)

> According to http://hennerley.nz/2018/01/29/g29-vs-g33/ 
good read :+1: 

SD 0.02 is pretty good, but you may have a warped alu bed (the cheap Chinese ones tend to be far from flat). Consider a round mirror or glass bed on top of it.

avri-schneider, 2018-05-03:
> Use G33 in the normal way.

What is the normal way? 

> You should set it before using G33 (or reissue a G33 command after each z-offset change)

Following G33, Z offset needs to be adjusted due to the Z-min probe being attached underneath the nozzle.

Detaching the probe and adjusting Z offset manually using a paper is required. Am I expected to issue a G33 without the probe attached? I assume it will slam into the bed or otherwise fail the Z probing.

If I'm expected to attach the probe and issue G33 again, wouldn't Z height be calibrated incorrectly by G33, as the probe is lower than the nozzle? Will it still work as intended?

> you may have a warped alu bed

The bed I have is very thick, I really doubt it is not flat, but I will try with a glass attached (already ordered a piece of round glass from my local glazier).

If it is indeed flat, and I experience the same behavior with the steps to reproduce reported in the issue - should this be considered a bug or expected behavior?

thinkyhead, 2018-05-04:
> wouldn't Z height be calibrated incorrectly by G33, as the probe is lower than the nozzle?

If your probe is lower than the nozzle, your `M851 Z` offset should be set to that distance. For example, `M851 Z-2.1` for a probe that triggers 2.1mm below the nozzle. Of course also set your X and Y probe offsets in whole mm.

LVD-AC, 2018-05-04:
> If your printer is reasonably well put together, and your print surface is actually flat...

If the bed is indeed flat it could be many things...

closed 2019-05-16 by thinkyhead