MarlinFirmware/Marlin #10608: SW SPI and HW SPI conflict

jacobsokolove, 2018-05-03:
### Description

I believe the SW SPI for my tmc 2130 and the HW spi for the sd card/lcd are conflicting. Basically, the printer recognizes the sd card and the files on the sd card, but when printing it, it homes and then waits there, homes, then the print is finished.

### Steps to Reproduce

1. Start Print via SD card
2. Printer Homes
3. Moves to print start position
4. homes again.
5. waits there until the job is "finished" (the length of the "job" is not remotely close to the expected print time. it says 3mins even when the test cube is printing, it should take at least 20 minutes).
6. rehomes. 

**print start ** [What you expect to happen]

** print "Starts" but never actually starts ** [What actually happens]
[Marlin for HC 300.zip](https://github.com/MarlinFirmware/Marlin/files/1969322/Marlin.for.HC.300.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-05-03:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

jacobsokolove, 2018-05-03:
I'm currently using the latest bugfix.

closed 2018-05-03 by jacobsokolove

Bob-the-Kuhn, 2018-05-03:
I looked at your configuration and it's pretty straight forward.  You're using AUX 2 for the TMC software SPI and AUX 1 for the TMC chip selects.

There are no SPI conflicts in your system.  There's actually three SPIs in your system:
- SD card hardware SPI on EXP2
- LCD software SPI on EXP1
- TMC2130 software SPI on AUX 1 & AUX 2

The default power up configuration for the TMC2130s has a high enough current that most steppers will move even if the TMC2130 isn't connected to a SPI.

Let's try an experiment.  Disconnect the TMC software SPI and see if the problem is the same.  If the steppers still move and the problem is the same then we need to look else where.  If the steppers just sit there and buzz then the default current is too low and we'll have to re-attach the TMC2130 SPI.

I suspect there is a problem either with your configuration or your slicer settings.  Please ZIP and attach the gcode file you're printing and a copy of the log.

thinkyhead, 2018-05-04:
…or, since you closed the issue, if you solved it, please let us know how!