MarlinFirmware/Marlin #10609: [2.0] FR PanelDue Integration

bwgeez02, 2018-05-03:
I have been chatting with dc42 about getting PanelDue support in Marlin. I'm no programmer/coder, so I'll just copy/paste his response. 

"PanelDue talks GCode over a serial interface. So the only thing you would need to do in Marlin to support is is to support the GCodes it uses, which are the ones with JSON responses. Specifically, M408 S0 and S1, M20 S2, and M36. You don't need to support all the M408 response fields because PanelDue will use whatever it finds."

"PanelDue already adjusts the GCodes it sends and what it displays when the host announces that it is Smoothieware or Repetier in the M408 response, to avoid using or displaying features not supported by those firmwares. We could do the same for Marlin if necessary."

"If you are serious about adding this support to Marlin, I can make a PanelDue display controller board available."

Thanks!

thinkyhead, 2018-05-03:
Seems like an interesting controller! I've added support at the above-linked pull request. If anyone wants to test the feature, simply connect the panel to an enabled serial port and enable `PANELDUE` in `Configuration.h`. Direct download of the branch is [here](https://github.com/thinkyhead/Marlin/archive/bf2_paneldue_json_M408.zip).

bwgeez02, 2018-05-03:
@thinkyhead , where do you want me to send a controller?

thinkyhead, 2018-05-04:
Hi @bwgeez02 — I look forward to being able to test the code on the actual device! Please send it to my mailing address, shown on this page: http://www.thinkyhead.com/support

bwgeez02, 2018-05-04:
Ordered one for me and one for you. USPS 2-3 day shipping. Can't wait! Thanks for all the contributions you and everyone else makes to the open source community!

thinkyhead, 2018-05-04:
Thanks! I look forward to spiffing it up. Our controller integration is pretty lacking right now.

thinkyhead, 2018-05-23:
Closing this, as it is now superseded by #10610.

closed 2018-05-23 by thinkyhead