MarlinFirmware/Marlin #10611: Q: Is stepper.synchronize needed in G92?

thinkyhead, 2018-05-03:
According to reports the typical `G92 E0` that slicers add before each layer lead to visible seams in the print due to the pause introduced by `planner.synchronize` at the start of `G92`.

The synchronize is included to ensure that the current planner/stepper positions aren't changed prematurely, while there are still moves in the planner, and it's pretty important —especially with XYZ positions— to make sure that the stepper/planner counts don't get out of phase. They need to be trustworthy for forward kinematics.

The E position is less important, since it isn't used for any form of forward kinematics, to restore the logical position based on the steppers. So it should be perfectly safe to skip `stepper.synchronize` when only updating the E axis. However, if E is changed at the moment that `G92` is executed, it will definitely be an untrustworthy value. When `planner.buffer_steps` is called (assuming E is sent to this method in absolute coordinates) it depends on the current planner E position to know how far to move.

So, the long and short of this is, we could drop the `stepper.synchronize` for changes in E only if the E coordinate sent to all planner move functions is guaranteed to be relative-only, so that there's no dependency on the E position in the planner. Then the E value tracked by the planner can be purely arbitrary.

Wot say?

CC: @MarlinFirmware/general-maintainers @MarlinFirmware/32bit-maintainers

GMagician, 2018-05-03:
Assumption: I may not know exactly how marlin is working at so low level hence I can write inappropriate thing.

I think that things will be easier acting like some industrial CNC works, using linked queues.
Tools change and G92 can store values as offset to current machine coordinates. When someone uses G92 or T then new offset are stored. When positions are required then proper offsets are added and then values are stored in planner. Using this approach only offset are changed when command are issued and planner doesn't care since it has only to execute its queued commands.

thinkyhead, 2018-05-03:
A question has come up about lost steps due to rounding down to the nearest step. Assuming an extruder has a typical 500 steps per mm, each lost step amounts to only 2 millimeters per meter of filament, or 0.2%. This is pretty negligible, but if it's very concerning, some steps can be made up by always adding 1/2 step before rounding, so there will be an equal number of rounding errors in the opposite direction on average.

closed 2018-05-04 by thinkyhead